agherzan/meta-raspberrypi #160: U-boot not working on rpi0w

keeslinp, 2017-11-22:
**Description**

I am trying to get mender to work on a rpi0-w and I am having trouble getting it to boot. It gets stuck at the rainbow screen which means that there is a problem with the kernel.img.

**Steps to reproduce the issue:**
1. Follow these steps https://github.com/mendersoftware/meta-mender/tree/master/meta-mender-raspberrypi

**Describe the results you received:**
rainbow screen

**Describe the results you expected:**
A successful boot

**Additional information you deem important (e.g. issue happens only occasionally):**

I found out that this is an issue with uboot and rpi0-w. https://raspberrypi.stackexchange.com/questions/68875/running-u-boot-on-a-pi-zero-w. <- this guy seems to have made a patch that works but I tried applying it and it did not work. It could be that I wasn't applying it correctly though. I also tested running mender on a rpi3 and it worked, so I don't think my configuration is the problem. I'm going to keep working with that patch and see if I can figure something out for my own situation. If I can fix it I'll see about doing a PR and we can work on a general solution.

betafive, 2017-11-22:
Can you ping the original patch author and see if he can update the patch and re-submit to the u-boot mailing list if it works?

keeslinp, 2017-11-22:
I'm actually not sure how to ping him, it seems I can't comment on the post because I have no reputation in that community. As I was poking around I had a thought about whether or not `make rpi_0_w_defconfig` is ever being called. Where are all the other `defconfig`s being `make`ed?

betafive, 2017-11-22:
The variable which determines the u-boot configuration used in OE is `UBOOT_MACHINE`. We define this in each machine's conf file under `conf/machine`. You should be able to change this in `raspberrypi0-wifi.conf` to match the newly introduced configuration.

keeslinp, 2017-11-22:
That helped! Now I was able to get past the rainbow screen and it seems to be stuck at "Starting kernel ..." From what I can see uboot found everything to be okay though. I'm going to try and take out mender and see if just normal uboot works.

keeslinp, 2017-11-22:
It still gets stuck at the same spot.
[image]

this is my local.conf:

``` 
CONF_VERSION = "1"
CORE_IMAGE_EXTRA_INSTALL += "openssh rsync sensorhub"
IMAGE_INSTALL_append += "nodejs nodejs-npm bluez5 networkmanager linux-firmware-bcm43430 config systemd"
IMAGE_INSTALL_append += " crda i2c-tools hostapd linux-firmware-bcm43430 linux-firmware-ralink linux-firmware-rtl8192ce linux-firmware-rtl8192cu linux-firmware-rtl8192su wireless-tools dhcp-client iw wpa-supplicant"
DISTRO_FEATURES_append += " bluez5 bluetooth wifi systemd"
RPI_USE_U_BOOT = "1"
INHERIT += "rm_work"
MACHINE = "raspberrypi0-wifi"
KERNEL_IMAGETYPE = "uImage"
```

betafive, 2017-11-22:
I'm guessing this is similar to the issue we had on 64-bit Raspberry Pi 3 devices with HDMI output. Sadly I can't help much as I don't have an R Pi Zero.

If you can get a non-OE/Yocto based image booting and outputting graphics (e.g. I know Raspbian, Arch Linux and OpenSUSE are usually good) then compare the `config.txt` file in the boot partition between that image and the image generated from OE/Yocto. You can also try overwriting just the firmware files or just the kernel in the OE/Yocto image with files from a working image to track down exactly which component is causing the issues. It's a bit slow and manual but it's pointed me in the right direction before.

keeslinp, 2017-11-22:
If I comment out `RPI_USE_U_BOOT = "1"` and `KERNEL_IMAGETYPE = "uImage"` then it boots up like normal and works fine. Would uBoot specifically mess up the hdmi?

I'll check out the config.txt and compare. thanks for the tip.

keeslinp, 2017-11-22:
The config.txt's are identical. 

This looks promising: http://processors.wiki.ti.com/index.php/Kernel_-_Common_Problems_Booting_Linux (problem 1). I'll see if I can fix that.

keeslinp, 2017-11-22:
I found this in the patch file

```
+         This option assumes the VideoCore firmware is configured to use the
+         mini UART (rather than PL011) for the serial console. This is the
+         default on the RPi Zero W. To enable the UART console, the following
+         non-default option must be present in config.txt: enable_uart=1.
+         This is required for U-Boot to operate correctly, even if you only
+         care about the HDMI/usbkbd console.
```

But I just tried adding that with no success.

keeslinp, 2017-11-27:
I did a comparison between what the pi3 and the pi0w look like while they boot. Here is an image of the pi3 booting:
[image]

The two big differences I noted were that the pi0w doesn't read any usb controllers but the pi3 does. Also, the pi3 seems to have a Flattened  Device Tree being applied. I'm not sure where to find that FDT but I have found some references to it so maybe that will lead somewhere.

marvic41, 2017-12-06:
Hello,

keeslinp, 2017-12-06:
We could try adding it as a patch like I was doing with the other code. If that works, let me know. My company decided to table mender for a bit and find another solution until uboot works on raspberry pi0w. If it does work I'm sure they'd love to know. Good luck :)

mirzak, 2017-12-06:
Well it it currently in upstream it will eventually end up here, next release of U-boot is scheduled for 2018.01 I believe and at which point it will be updated in oe-core and we will have support for it.

Until then above patches could be integrated in meta-raspberrypi (bbappend) with status "Backport" until 2018.01 is released. Whoever has interest in having this support please send patches :)

Example on how it is done is [1] and then removed in [2].

[1]. https://github.com/agherzan/meta-raspberrypi/commit/94e2929f746f7e49a7870f7ea889dcbed05296c7#diff-81ef3473278e74e6b07dcfdac23cb8cc
[2]. https://github.com/agherzan/meta-raspberrypi/commit/5a2b34e56c228a2fa768a2b3feeebc9f4600b7c4#diff-81ef3473278e74e6b07dcfdac23cb8cc

marvic41, 2017-12-07:
Patched (from above link) at this ~/poky/meta/recipes-bsp/u-boot, but yocto not compiling. Than,
Cross compiled uboot (add support for Raspberry Pi Zero W) patched. moved uboot.bin to kernel.img
and copied to boot folder. It got booted with U-boot> prompt 
Some thing gone wrong.
[image]

mirzak, 2017-12-07:
You need to provide a boot file in some form.

When you build with you Yocto it creates a "boot.scr" file [1] and that is it located on the boot part (same partition as kernel7.img). If you cross-compiled U-boot you need to also create this file and put it in the correct location with the correct name (boot.scr).

[1]. https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bb

marvic41, 2017-12-07:
Patched (from above link) to ~/poky/meta/recipes-bsp/u-boot, yocto  did not compiled, than took the uboot (add support for Raspberry Pi Zero W ) source and  cross compiled. Changed uboot.bin to kernel.img
and replaced with old kernel.img in the boot partition. It got missing environment variable bootfile  errors, wrong with uboot source? 
[image]

marvic41, 2017-12-07:
boot.scr is not created after cross compiling U-boot, it created these files.
[image]

mirzak, 2017-12-07:
You need to manually do it, it is not part of the u-boot build process.

Please take a look at this recipe for inspiration (this is how it is done in Yocto), https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bb

marvic41, 2017-12-07:
I will try that. Patch (https://patchwork.ozlabs.org/patch/841439/) 
~/poky/meta/recipes-bsp/u-boot/u-boot-common_2017.01.inc   this giving errors. 
SRC_URI = "git://git.denx.de/u-boot.git \
file://001-uboot-zero-w.patch
"
What is way to do this?

mirzak, 2017-12-07:
Try this branch: https://github.com/mirzak/meta-raspberrypi/tree/rpi-zero-wifi-support

Compile tested it if you can "runtime" test it i will create an pull-request here.

marvic41, 2017-12-07:
I have to replace this branch git clone git://git.yoctoproject.org/meta-raspberrypi (current branch)
with yours
https://github.com/mirzak/meta-raspberrypi/tree/rpi-zero-wifi-support

mirzak, 2017-12-07:
Yepp

marvic41, 2017-12-08:
This branch rpi-zero-wifi-support also struck ed at rain bow screen, Patch for this branch is needed?

mirzak, 2017-12-08:
Only 

    RPI_USE_U_BOOT = "1"

is necessary.

Oh saw now that raspberrypi0-wifi machine conf defines UBOOT_MACHINE = "rpi_defconfig", which is wrong.

I have updated my branch with the correct value. [1].

[1]. https://github.com/mirzak/meta-raspberrypi/commits/rpi-zero-wifi-support

marvic41, 2017-12-08:
I took fresh copy of meta-raspberrypi branch rpi-zero-wifi-support and compiled it got this error.

`ERROR: oe_runmake failed
| /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.01-r0/git/scripts/Makefile.build:280: recipe for target 'common/lcd_simplefb.o' failed
| make[2]: *** [common/lcd_simplefb.o] Error 1
| make[2]: *** Waiting for unfinished jobs....
| /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.01-r0/git/Makefile:1217: recipe for target 'common' failed
| make[1]: *** [common] Error 2
| make[1]: Leaving directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.01-r0/build'
| Makefile:150: recipe for target 'sub-make' failed
| make: *** [sub-make] Error 2
| make: Leaving directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.01-r0/git'
| ERROR: Function failed: do_compile (log file is located at /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.01-r0/temp/log.do_compile.16293)
ERROR: Task (/home/kkk/poky/meta/recipes-bsp/u-boot/u-boot_2017.01.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3424 tasks of which 3422 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/kkk/poky/meta/recipes-bsp/u-boot/u-boot_2017.01.bb:do_compile
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
`

mirzak, 2017-12-08:
Which branch are you using for poky?

marvic41, 2017-12-08:
Tried with both, it got compiled with y'days update. With latest update (today) meta-raspberrypi got error
git clone -b pyro git://git.yoctoproject.org/poky

git clone -b master git://git.yoctoproject.org/poky

mirzak, 2017-12-08:
It will unlikely work with pyro as probably requires some more "backported" patches.

Just did a compile test and it passed. My conf:

```
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.9"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4+snapshot-20171208"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:4469acdf1d0338220f3fe2ecb5e079eea6fda375"
meta-multimedia      
meta-oe              
meta-networking      
meta-python          = "master:e08670d4362b57373557fbe6b31cbffffa465acc"
meta-raspberrypi     = "rpi-zero-wifi-support:c3d343888b278e1c4f26e32d8fb24110e957217b"
```

marvic41, 2017-12-08:
This the output
kkk@mender178:~/poky/build$ bitbake rpi-hwup-image
Loading cache: 100% |############################################| Time: 0:00:02
Loaded 2863 entries from dependency cache.
Parsing recipes: 100% |##########################################| Time: 0:00:03
Parsing of 2043 .bb files complete (2041 cached, 2 parsed). 2866 targets, 106 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION                                = "1.34.0"
BUILD_SYS                                   = "x86_64-linux"
NATIVELSBSTRING                       = "universal"
TARGET_SYS                                 = "arm-poky-linux-gnueabi"
MACHINE                                     = "raspberrypi0-wifi"
DISTRO                                         = "poky"
DISTRO_VERSION                          = "2.3.2"
TUNE_FEATURES                            = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU                                  = "hard"
meta
meta-poky
meta-yocto-bsp                             = "pyro:2021e95bcd986e5bcfe457da5c3e16d60241d050"
meta-mender-core
meta-mender-raspberrypi              = "pyro:b40f05c82ebc97c18ddf8ac426af4657ddba9425"
meta-raspberrypi  = "rpi-zero-wifi-support:5c1daa4ecb9ae608100d35daf85d38cdc0e20d2f"
meta-oe
meta-multimedia
meta-python
meta-networking   = "master:e08670d4362b57373557fbe6b31cbffffa465acc"

marvic41, 2017-12-08:
Where to find  backported patches?

mirzak, 2017-12-08:
You would have to create them your self based on upstream sources.

marvic41, 2017-12-08:
Any other poky branch will support without patches ?

marvic41, 2017-12-08:
Trying to build from master poky branch, got this erros
`kkk@mender178:~/poky/build$ bitbake-layers add-layer ../meta-mender/meta-mender-raspberrypi
NOTE: Starting bitbake server...
Parsing recipes: 100% |##########################################| Time: 0:00:37
Parsing of 830 .bb files complete (0 cached, 830 parsed). 1293 targets, 43 skipped, 0 masked, 0 errors.
ERROR: No recipes available for:
  /home/kkk/poky/meta-mender/meta-mender-raspberrypi/recipes-bsp/rpi-u-boot-scr/rpi-u-boot-scr.bbappend
  /home/kkk/poky/meta-mender/meta-mender-raspberrypi/recipes-kernel/linux/linux-raspberrypi_%.bbappend
`

marvic41, 2017-12-08:
Dependencies need to add first,  now error gone,

marvic41, 2017-12-08:
Even on got the same error for poky master branch, after adding this branch
git clone https://github.com/mirzak/meta-raspberrypi/ -b rpi-zero-wifi-support

No errors with this branch

mirzak, 2017-12-08:
Can you post your full configuration (what is printed when you run bitbake)

marvic41, 2017-12-08:
This after bitbake rpi-hwup-image
------------
kkk@mender178:~/poky/build$ bitbake rpi-hwup-image
Parsing recipes: 100% |##########################################| Time: 0:02:07
Parsing of 2024 .bb files complete (0 cached, 2024 parsed). 2833 targets, 99 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4+snapshot-20171208"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "master:4469acdf1d0338220f3fe2ecb5e079eea6fda375"
meta-mender-core     = "master:09b88623ea1360bc2cfd2bc57a237a14bbeb73e0"
meta-oe
meta-multimedia
meta-python
meta-networking      = "master:e08670d4362b57373557fbe6b31cbffffa465acc"
meta-raspberrypi     = "rpi-zero-wifi-support:c3d343888b278e1c4f26e32d8fb24110e957217b"
meta-mender-raspberrypi = "master:09b88623ea1360bc2cfd2bc57a237a14bbeb73e0"

Initialising tasks: 100% |#######################################| Time: 0:00:49
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: u-boot-1_2017.11-r0 do_compile: oe_runmake failed
ERROR: u-boot-1_2017.11-r0 do_compile: Function failed: do_compile (log file is located at /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/temp/log.do_compile.1260)
ERROR: Logfile of failure stored in: /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/temp/log.do_compile.1260
Log data follows:
| DEBUG: Executing shell function do_compile
| NOTE: make -j 8 CROSS_COMPILE=arm-poky-linux-gnueabi- CC=arm-poky-linux-gnueabi-gcc  --sysroot=/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot V=1 HOSTCC=gcc  -isystem/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/lib -L/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/lib -Wl,-rpath,/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/lib -Wl,-O1 -C /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git O=/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/build rpi_0_w_defconfig
| make: Entering directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git'
| make -C /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/build KBUILD_SRC=/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git \
| -f /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git/Makefile rpi_0_w_defconfig
| make[1]: Entering directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/build'
| make -f /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git/scripts/Makefile.build obj=scripts/basic
| ln -fsn /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git source
| rm -f .tmp_quiet_recordmcount
| /bin/bash /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git/scripts/mkmakefile \
|     /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git . 2017 11
|   GEN     ./Makefile
| make -f /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git/scripts/Makefile.build obj=scripts/kconfig rpi_0_w_defconfig
| scripts/kconfig/conf  --defconfig=arch/../configs/rpi_0_w_defconfig Kconfig
| #
| # configuration written to .config
| #
| make[1]: Leaving directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/build'
| NOTE: make -j 8 CROSS_COMPILE=arm-poky-linux-gnueabi- CC=arm-poky-linux-gnueabi-gcc  --sysroot=/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot V=1 HOSTCC=gcc  -isystem/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/lib -L/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/lib -Wl,-rpath-link,/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/usr/lib -Wl,-rpath-link,/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/recipe-sysroot-native/lib -Wl,-rpath,/home/kkk/poky/bui [... cut ...]
make[1]: Leaving directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/build'
ERROR: oe_runmake failed
Makefile:150: recipe for target 'sub-make' failed
make: *** [sub-make] Error 2
make: Leaving directory '/home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0/git'
WARNING: exit code 1 from a shell command.
ERROR: Function failed: do_compile (log file is located at /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.11-r0$

.

mirzak, 2017-12-08:
Can you fix the formatting? Very hard to read now.

Anyway, you are using meta-mender-core which appends to u-boot. Please for the sake of what we are trying to accomplish (u-boot support on raspberrypi0-wifi) remove the meta-mender-core layers and build a "stock" raspberrypi image.

The errors related to Mender you should bring to the Mender mailing-list.

marvic41, 2017-12-08:
Without meta-mender core layer it will build, Already it is done.
But, mender with RPi zero W is needed.
With this RPi branch git://git.yoctoproject.org/meta-raspberrypi
Image created, but, it struck ed at rain bow screen. Added boot_delay=1 to config file in boot partition. increased to 20. Still struck t rain bow display.

marvic41, 2017-12-08:
After changing from 
UBOOT_MACHINE ?= "rpi_config"
to
+UBOOT_MACHINE ?= "rpi_0_w_defconfig"  it is flashing error here

scripts/kconfig/conf  --defconfig=arch/../configs/rpi_0_w_defconfig Kconfig
| ***
| *** **Can't find default configuration "arch/../configs/rpi_0_w_defconfxig"!**
| ***
| /home/kkk/poky/build/tmp/work/raspberrypi0_wifi-poky-linux-gnueabi/u-boot/1_2017.01-r0/git/scripts/kconfig/Makefile:114: recipe for target 'rpi_0_w_defconfig' failed

ERROR: Task (/home/kkk/poky/meta/recipes-bsp/u-boot/u-boot_2017.01.bb:do_compile) failed with exit code '1'
NOTE: Tasks Summary: Attempted 3424 tasks of which 3417 didn't need to be rerun and 1 failed.

This means this main line (git://git.denx.de/u-boot.git) is not updated ?

marvic41, 2017-12-09:
I took fresh copy of all and compiled without mender, it compiled properly.
Next added git://github.com/mendersoftware/meta-mender layer. Inside this folder 
https://github.com/mendersoftware/meta-mender/blob/master/meta-mender-raspberrypi/recipes-bsp/u-boot/patches/0001-CONFIGS-rpi-enable-mender-requirements.patch
this patch doen't have for  rpi_0_w_defconfig, added this to patch

configs/rpi_0_w_defconfig   | 1 +
diff --git a/configs/rpi_0_w_defconfig b/configs/rpi_0_w_defconfig
 index 2345697..ef3c0d8 100644
 --- a/configs/rpi_0_w_defconfig
 +++ b/configs/rpi_0_w_defconfig
@@ -28,3 +28,4 @@ CONFIG_SYS_WHITE_ON_BLACK=y
 CONFIG_CONSOLE_SCROLL_LINES=10
 CONFIG_PHYS_TO_BUS=y
CONFIG_OF_LIBFDT_OVERLAY=y
+CONFIG_ENV_IS_IN_MMC=y

Error 
Applying patch 0001-CONFIGS-rpi-enable-mender-requirements.patch
can't find file to patch at input line 26
Perhaps you used the wrong -p or --strip option?
No file to patch.  Skipping patch.

Is rpi_0_w_defconfig file missing in U-boot folder?

mirzak, 2017-12-10:
@marvic41 can you confirm for me that U-boot now works for you your raspberrypi0-wifi when you only build with meta-raspberrypi?

If that is the case we can close this issue and I can send a pull-request to get the support up-streamed.

The error you are experiencing now is meta-mender related and we can move that discussion there (I will pop up there as well).

marvic41, 2017-12-11:
It got booted mata-raspberrypi for wifi without mender. This worked before without patch for Zero w Pi.

If this is commented RPI_USE_U_BOOT = "1".

mark2b, 2018-02-21:
Are you going to merge RPI Zero W U-Boot fix to Rocko?

betafive, 2018-02-26:
I'm not planning on backporting this to rocko as it's not a major bugfix or security issue and I'm a bit limited on time right now. I'd accept a patch for rocko which applies changes to u-boot only when `MACHINE=raspberrypi0-wifi` though.

closed 2018-02-26 by betafive