agherzan/meta-raspberrypi #162: wget the rpio fail

wuxx, 2017-11-24:
**Description**
when I use this command  to compile image for my rpi 3B
$bitbake qt5-image
the bitbake report the wget rpio fail and I have found the reason and the solution.

the origin log output have be overwritten, but we can reproduce it, just use the wget cmd

**Steps to reproduce the issue:**
1.
$wget http://pypi.python.org/packages/source/R/RPIO/RPIO-0.10.0.tar.gz
--2017-11-24 18:10:22--  http://pypi.python.org/packages/source/R/RPIO/RPIO-0.10.0.tar.gz
Resolving pypi.python.org (pypi.python.org)... 151.101.72.223, 2a04:4e42:11::223
Connecting to pypi.python.org (pypi.python.org)|151.101.72.223|:80... connected.
HTTP request sent, awaiting response... 403 SSL is required
2017-11-24 18:10:27 ERROR 403: SSL is required.
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
after I add the 's' after the "http", to get the package with ssl protocol, problem solved, so I think the script meta-raspberrypi/recipes-devtools/python/rpio_0.10.0.bb line 10 need to be update.

wuxx, 2017-11-24:
added: I work with the branch rocko

betafive, 2017-11-27:
It should be possible to modify the recipe to use the pypi class from the meta-python layer. This will set SRC_URI for us from the package name.

Would you like to take a look at that and send a patch?

betafive, 2017-11-27:
Duplicates #146

closed 2017-11-27 by betafive