agherzan/meta-raspberrypi #169: Implement  RPI_EXTRA_CONFIG

betafive, 2017-12-04:
(putting this in here so I don't forget)

It would be good to have an RPI_EXTRA_CONFIG variable or similar to add arbitrary text to the config file, so you could do something like this:

```
RPI_EXTRA_CONFIG = " \
    # Raspberry Pi 7\" display/touch screen \
    lcd_rotate=2 \
    "
```

closed 2017-12-22 by betafive

twoerner, 2018-01-09:
Does this work for others? It doesn't seem to work for me.

RPI_EXTRA_CONFIG = " \n \
    dtoverlay=spi1-3cs \n \
    dtoverlay=fti-tpm \n \
    "

gives a /mnt/config.txt that looks like:

...
#dtparam=pwr_led_gpio=35
# Enable SPI bus
dtparam=spi=on
# Enable I2C
dtparam=i2c1=on
dtparam=i2c_arm=on
# Enable UART
enable_uart=1
 \n dtoverlay=spi1-3cs \n dtoverlay=fti-tpm \n

betafive, 2018-01-11:
@twoerner: I've just re-tested this in master with your lines added to my local.conf.

Output looks like this:

```
$ cat tmp/deploy/images/raspberrypi3/bcm2835-bootfiles/config.txt

<snip>

#dtparam=pwr_led_gpio=35

 dtoverlay=spi1-3cs
 dtoverlay=fti-tpm

```

Not sure why this would be working for me but not for you. Are you using latest master of everything?

agherzan, 2018-01-11:
Test it too and looks fine.

memox5, 2019-01-15:
using "echo -e" seems to fix the issue.
tested on ubuntu 16.04

> 
> 
> Does this work for others? It doesn't seem to work for me. The following in local.conf:
> 
> ```
> RPI_EXTRA_CONFIG = " \n \
>     dtoverlay=spi1-3cs \n \
>     dtoverlay=fti-tpm \n \
>     "
> ```
> 
> gives a /mnt/config.txt that looks like:
> 
> ```
> ...
> #dtparam=pwr_led_gpio=35
> # Enable SPI bus
> dtparam=spi=on
> # Enable I2C
> dtparam=i2c1=on
> dtparam=i2c_arm=on
> # Enable UART
> enable_uart=1
>  \n dtoverlay=spi1-3cs \n dtoverlay=fti-tpm \n
> ```

using echo -e seems to fix the issue, tested on ubuntu 16.04

agherzan, 2019-01-15:
Sounds like a bug @memox5. Would you be able to push a fix? Thanks in advance.

memox5, 2019-01-16:
#370

memox5, 2019-01-16:
#370 should do

memox5, 2019-01-16:
@agherzan #370 should do, thank you.