agherzan/meta-raspberrypi #172: "You had no conf/local.conf file." message upon

kamomil, 2017-12-12:
**Description**

**Steps to reproduce the issue:**
1.
2.
3.

**Describe the results you received:**

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

mirzak, 2017-12-12:
This is expected behavior when you create an a new "build" directory. The `local.conf` file is simply a copy of a template file found in poky [1]. 

Just set MACHINE to desired raspberrypi board.

[1]. http://git.yoctoproject.org/cgit.cgi/poky/tree/meta-poky/conf/local.conf.sample?h=rocko

kamomil, 2017-12-13:
Hi,
I set `MACHINE = raspberrypi2`, then running `bitbake rpi-hwup-image` I got the error:

```
ERROR:  OE-core's config sanity checker detected a potential misconfiguration.
    Either fix the cause of this error or at your own risk disable the checker (see sanity.conf).
    Following is the list of potential problems / advisories:

    MACHINE=raspberrypi2 is invalid. Please set a valid MACHINE in your local.conf, environment or other configuration file.
```

mirzak, 2017-12-13:
You need to add meta-raspberrypi to your bblayers.conf

agherzan, 2017-12-13:
This is not a bug and the problems you encounter are basic yocto concepts that you need to handle before being able to work with this BSP layer. I suggest reading http://www.yoctoproject.org/docs/current/yocto-project-qs/yocto-project-qs.html and as well take a look as much as possible at http://www.yoctoproject.org/docs/current/ref-manual/ref-manual.html .

closed 2017-12-13 by agherzan