agherzan/meta-raspberrypi #173: No network with pi3

laibulle, 2017-12-13:
Hello, 

I am trying to get network on my Raspberry Pi3.

`ip link` returns "lo" and "eth0" (where is wlan0 ?)

but ifup returns
```
ifup eth0 
ifup: can't open '/etc/network/interface': No such file or directory

```
Also, before the boot I get "No ethernet found" and the ethernet LED are off until the linux boot.

Did a miss something in the configuration ?

Best regards.

local.conf

```
MENDER_ARTIFACT_NAME = "release-1"

INHERIT += "mender-full"
MACHINE = "raspberrypi3"

RPI_USE_U_BOOT = "1"
MENDER_PARTITION_ALIGNMENT_KB = "4096"
MENDER_BOOT_PART_SIZE_MB = "40"
IMAGE_INSTALL_append = " kernel-image kernel-devicetree"
IMAGE_FSTYPES_remove += " rpi-sdimg"
IMAGE_BOOT_FILES_append = " boot.scr u-boot.bin;${SDIMG_KERNELIMAGE}"
KERNEL_IMAGETYPE = "uImage"

MENDER_SERVER_URL = "https://hosted.mender.io"
MENDER_TENANT_TOKEN = "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx"

DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""

IMAGE_FSTYPES = "ext4"

PACKAGE_CLASSES ?= "package_rpm"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"

BB_DISKMON_DIRS = "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

PACKAGECONFIG_append_pn-qemu-native = " sdl"
PACKAGECONFIG_append_pn-nativesdk-qemu = " sdl"
CONF_VERSION = "1"

```
bblayer.conf

```
# POKY_BBLAYERS_CONF_VERSION is increased each time build/conf/bblayers.conf
# changes incompatibly
POKY_BBLAYERS_CONF_VERSION = "2"

BBPATH = "${TOPDIR}"
BBFILES ?= ""

BBLAYERS ?= " \
  /home/laibulle/poky/meta \
  /home/laibulle/poky/meta-poky \
  /home/laibulle/poky/meta-yocto-bsp \
  /home/laibulle/poky/meta-haum \
  /home/laibulle/poky/meta-nodejs \
  /home/laibulle/poky/meta-raspberrypi \
  /home/laibulle/poky/meta-mender/meta-mender-core \
  /home/laibulle/poky/meta-mender/meta-mender-raspberrypi \
  "
```

agherzan, 2017-12-13:
@laibulle What image do you install and can you pastebin dmesg?

jerbob92, 2017-12-16:
Wifi isn't working here either. I added linux-firmware-bcm43430 to IMAGE_INSTALL_append.

agherzan, 2017-12-21:
It does work with the right firmware added. This layer come with some images for testing. The key thing is https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-core/packagegroups/packagegroup-rpi-test.bb#L31 .

closed 2017-12-21 by agherzan

jerbob92, 2017-12-21:
@agherzan can you tell me which layers I should add in my local.conf?
Because I'm not getting it to work, whatever I try.

jerbob92, 2017-12-21:
Thanks, already found kernel-modules in rpi-hwup-image so building that now. linux-firmware-bcm43430 was already in there. Will let you know.

agherzan, 2017-12-21:
But generally use the test image. And start from there when investigating support issues.