agherzan/meta-raspberrypi #174: hciattach is faulty

sherifomran, 2017-12-18:
**Description**

**Steps to reproduce the issue:**
1. download raspbian image
2. create image using yocto for rpi0w
3. hciattach /dev/ttyAMA0 bcm43xx 3000000 (using the raspian and yocto images)

**Describe the results you received:**
with yocto image, there is initialization timeout. But with the raspbian image, it works fine. I copied the hciattach + the driver to the yocto image (manuall) and it worked. 

**Describe the results you expected:**
hciconfig -a should give a reply that the chip exist

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

sherifomran, 2017-12-18:
I could fix get it to work for the raspberry pi 0w.
The driver BCM43430A1.hcd 360kb is not correct. The raspbian image has another driver. I took it and embedded it in a recipie to replace this driver. Also, in bluez5.%.bbappend, the patches are installed for raspberrypi3 (see file: SRC_URI_append_raspberrypi3) remove  all existances or "_raspberrypi3" to apply the patch also for the rpi0w.

now by giving hciattach /dev/ttyAMA0 bcm43xx 3000000 it works .. Device setup complete ..

agherzan, 2017-12-21:
Actually we do it for wifi0 too. Check the recipe in master. Would you be able to update BCM43430A1.hcd on master?

agherzan, 2017-12-21:
Closing as it duplicates https://github.com/agherzan/meta-raspberrypi/issues/148

sherifomran, 2017-12-21:
I have the morty branch, please update the bluez5% also to it. I sent you the driver per email. thank you

agherzan, 2017-12-21:
Please send it over as a PR to master. When merged you can send another one which backports it over morty.

agherzan, 2018-01-22:
Closing as duplicate of #148

closed 2018-01-22 by agherzan