agherzan/meta-raspberrypi #177: U-Boot not working with Rocko on Raspberry Pi 3

leon-anavi, 2018-01-10:
**Description**

Image rpi-basic-image fails to boot on Raspberry Pi 3 using U-Boot and recipes from Poky and meta-raspberrypi branches Rocko. The board gets stuck at the rainbow screen. There is no output on the UART. 

I wondering if this issue could be related to #160.

**Steps to reproduce the issue:**
1. Checkout Poky branch Rocko as well as meta-raspberrypi branch Rocko.
2. Set the following configurations in conf/local.conf:
```
DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""

ENABLE_UART = "1"

KERNEL_IMAGETYPE = "uImage"
PREFERRED_PROVIDER_virtual/bootloader ?= "u-boot"
UBOOT_MACHINE_raspberrypi3 ?= "rpi_3_32b_defconfig"
```
3. Build and try to boot **rpi-basic-image** for **raspberrypi3**

**Describe the results you received:**
Raspberry Pi 3 doesn't boot. The board gets stuck at the rainbow screen. No output on the UART. 

**Describe the results you expected:**
The expected result is Raspberry Pi 3 to boot successfully.

**Additional information you deem important (e.g. issue happens only occasionally):**
The issues happens all the time with the described steps to reproduce it.

**Additional details (revisions used, host distro, etc.):**
My build configuration is:
```
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "ubuntu-16.04"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4.1"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
```

Best regards, Leon

betafive, 2018-01-10:
The option to enable u-boot changed with the rocko release (to allow us to support different kernel image types with u-boot, needed for rpi 3 64-bit u-boot).

The short answer is you now need to set `RPI_USE_U_BOOT = "1"`.

See https://github.com/agherzan/meta-raspberrypi/blob/rocko/docs/extra-build-config.md#boot-to-u-boot for the updated documentation.

leon-anavi, 2018-01-10:
@betafive, thank you the quick feedback. Works like a charm. 

Thanks, Leon

closed 2018-01-10 by leon-anavi

mfrydrysiak, 2018-02-08:
Hi,
I'm facing a similar issue with my RPi3. I use the Rocko branches on meta-raspberrypi, meta-oe and poky. The `conf/local.conf` file contains:
```
ENABLE_UART = "1"
KERNEL_IMAGETYPE = "uImage"
PREFERRED_PROVIDER_virtual/bootloader ?= "u-boot"
UBOOT_MACHINE_raspberrypi3 ?= "rpi_3_32b_defconfig"
RPI_USE_U_BOOT = "1"
MACHINE = "raspberrypi3"
```
I built the image with `bitbake rpi-hwup-image`.

After connecting the RPi3 through the UART/USB FTDI converter I do not get any prints on the console (checked also the baudrate = 115200). If I boot the board with the HDMI cable connected I can see the U-Boot prompt, but none of the bootloader commands work (such as pr, bdinfo, setenv, etc.).