agherzan/meta-raspberrypi #179: ttyAMA0 from app not working

AntonioSanta, 2018-01-17:
**Description**
I cannot drive the UART0_TX (pin#8) from cpp application. The port ttyAMA0 can be opened successfully from the app but when trying writing some data nothing happens on the TX pin.  

**Steps to reproduce the issue:**
1. set ENABLE_UART = "1" and  MACHINE_FEATURES += "pi3-disable-bt" . Changed the SERIAL_CONSOLE on the raspberrypi3.conf file to SERIAL_CONSOLE ?= "115200 tty1" followed the instructions on the docs http://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#enable-uart
2.rebuild the image nad load it to the sd card.
3.when raspberrypi3 boots up check with a simple app or echo '1' > /dev/ttyAMA0 , nothing is output to the tx pin of the UART ttyAMA0.

**Describe the results you received:**
the TX pin does not move, nothing is transmitted and no error is reported.

**Describe the results you expected:**
the TX pin to transmit the data

**Additional information you deem important (e.g. issue happens only occasionally):**
the same application on the same hw is working using Raspbian original image.
**Additional details (revisions used, host distro, etc.):**
it seems the console is retained anyway to ttyAMA0 as dmesg | grep ttyAMA0  output shows kernel command line: ...... console=ttyAMA0,115200. Then how to change the console to go to another uart ?

agherzan, 2018-01-22:
What is the RPi model you are working on?

AntonioSanta, 2018-01-22:
hello, it is raspberryPi3 model B , I have made the UART0_TX pin working by using the ttyS0 port instead of ttyAMA0.

agherzan, 2018-01-22:
Can we close the issue in that case? @AntonioSanta

AntonioSanta, 2018-01-24:
OK with me if this is as expected.

agherzan, 2018-02-05:
@AntonioSanta It sounds like an issue. Would you be able to try to reproduce on a raspbian?

AntonioSanta, 2018-02-15:
On Raspbian 8 ( jessie )(with  Linux 4.9.35-v7 ) the same application is successfully driving the pin UART0_TX pin when opening ttyAMA0 device.
On the OS built with meta-raspberrypi instead the pin is successfully driven when using the ttyS0 device from the app. When using ttyAM0 the port can be opened but when writing to the tx nothing happens to the UART0_TX pin.

agherzan, 2018-03-01:
@AntonioSanta are you able to give it a try with the current master (we update firmware and kernel)?

61131, 2018-04-05:
Known behaviour with RP3 - See https://github.com/raspberrypi/linux/issues/1399

agherzan, 2018-04-05:
@61131 But the pi3-disable-bt should switch roles. That's the issue reported here.

agherzan, 2018-07-26:
By all means, this sounds like an issue in the kernel. I recommend trying newer meta-raspberrypi revisions to reproduce. Otherwise, follow it up in linux-raspberrypi github repository. Closing for now.

closed 2018-07-26 by agherzan

AntonioSanta, 2018-07-26:
I am upgrading now the system to the newer Sumo version. I couldn't check
before.

agherzan, 2018-07-26:
Looking forward to hearing your findings. 
-- 
Andrei Gherzan
gpg: rsa4096/D4D94F67AD0E9640 | t: @agherzan

AntonioSanta, 2018-08-08:
I checked with the Sumo version and the behavior is still the same ,it works with ttyS0. I will follow on linux-raspberrypi, as you said.