agherzan/meta-raspberrypi #187: kernel-devsrc fails to install, broken dependencies

fgiancane8, 2018-01-31:
Hi all,

For some reasons, I had to build a Raspberry Pi 3 image with kernel sources in both SDK and image.

After building the package (a simple patch is needed, but the package builds correctly) the do_install step fails, because the kernel-devsrc package imports the dependencies from virtual/kernel and in turn from linux-raspberrypi.

The failure is in the fact that a requirement is written into the .spec RPM file, which is upon '/bin/awk' which is non-existent.

Given that using a different machine, like generic-x86_64 this problem does not arise, I think that it is bound to a particular issue with raspberry pi machine(s).

I looked at the recipes and the meta-raspberrypi layer but could not find anything useful.

**Steps to reproduce the issue:**
1. create a bbappend file for any rpi image
2. add 'kernel-devsrc' to IMAGE_INSTALL
3. bitbake the image
4. see results

**Describe the results you received:**
ERROR: rpi-sdk-image-1.0-r0 do_populate_sdk: Could not invoke dnf. Command '/home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/recipe-sysroot-native/usr/bin/dnf -y -c /home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/sdk/image/opt/clik/2.4.1/sysroots/aarch64-poky-linux/etc/dnf/dnf.conf --setopt=reposdir=/home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/sdk/image/opt/clik/2.4.1/sysroots/aarch64-poky-linux/etc/yum.repos.d --repofrompath=oe-repo,/home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/oe-sdk-repo --installroot=/home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/sdk/image/opt/clik/2.4.1/sysroots/aarch64-poky-linux --setopt=logdir=/home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/temp --nogpgcheck install alsa-plugins alsa-utils avahi-daemon avahi-dnsconfd connman connman-client connman-wait-online kernel-devsrc kernel-modules linux-firmware ntp packagegroup-core-boot packagegroup-core-sdk packagegroup-core-ssh-dropbear packagegroup-core-standalone-sdk-target psplash-raspberrypi run-postinsts screen' returned 1:
Added oe-repo repo from /home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/oe-sdk-repo
Last metadata expiration check: 0:00:00 ago on Sat 27 Jan 2018 07:12:15 PM UTC.
Error:
 Problem: conflicting requests
  - nothing provides /bin/awk needed by kernel-devsrc-1.0-r0.raspberrypi3_64

ERROR: rpi-sdk-image-1.0-r0 do_populate_sdk: Function failed: do_populate_sdk
ERROR: Logfile of failure stored in: /home/fgp/poky/rpi3_64/tmp/work/raspberrypi3_64-poky-linux/rpi-sdk-image/1.0-r0/temp/log.do_populate_sdk.20465
ERROR: Task (/home/fgp/poky/meta-clik/recipes-clik/images/rpi-sdk-image.bb:do_populate_sdk) failed with exit code '1'

Working on rocko and fedora 27

agherzan, 2018-03-01:
In order to fox this should would need to apply a patch like this. http://git.yoctoproject.org/cgit/cgit.cgi/linux-yocto/commit/?id=8af11c1cdd8fa08217e702b57cf96e9030db52b2 . Would you be able to submit it as a PR?

agherzan, 2018-03-01:
I checked the upstream and it seems like this was fixed in kernel 4.12+. We updated master to 4.14 so you should be able to get this fixed by switching (updating) to master. It would be nice as well if you could backport this patch for the 4.9 recipe as well.

fgiancane8, 2018-03-02:
Got it. Thanks for the comment.

Will do a branch and a PR as soon as possible.

agherzan, 2018-03-05:
Thanks @fgiancane8

kraj, 2018-04-16:
@fgiancane8 is this still on your TODO ?

fgiancane8, 2018-05-02:
Hi @kraj ,

Have been very busy with my job lately. If this is not already done, I will arrange to create a branch and a PR as soon as possible.

Thanks,
F

manszigher, 2018-10-25:
Any solution on this? I am facing the same issue but on another target. Even though the patch will fix the problem so that we can run kernel-devsrc the problem persist for do_populate_sdk which will still fail because rpm depends on /bin/awk for some reason that I have not figured out.

kraj, 2018-10-25:
that could be something else needing awk and doing something similar. You might want to grep for /bin/awk in rootfs directory inside image bb temp build area.

manszigher, 2018-10-26:
So adding this to kernel-devsrc.bbappend solved my problem

do_install_append() {
    # This fixes the rpm dependency failure on install of kernel-devsrc depending on /bin/awk
    cd ${D} || true
    for i in $(grep -srI "\!/bin/awk" | cut -d":" -f1); do
        sed -i -e "s#\!/bin/awk#\!/usr/bin/env awk#g" $i
    done
}

manszigher, 2018-10-26:
And here is the original solution

https://gerrit.automotivelinux.org/gerrit/gitweb?p=AGL/meta-agl.git;a=blobdiff;f=meta-agl-bsp/meta-ti/recipes-kernel/linux/linux-ti-staging_%25.bbappend;h=43fa494a26c132b875c177acf0389163d5e34c06;hp=b2e5431400acef0a5372e5490ac4f464482d4b3b;hb=4bfc821810cdee47611c6d3e94d771971f51fa75;hpb=1cf6b17fd15fde569374c85d4df722784f4e9c67

manszigher, 2018-10-26:
No sure that the patch for the kernel is needed or not anymore but maybe there should be more patches on the kernel to remove all /bin/awk in the scripts. But appending on kernel-devsrc is at least a solution that works without having to patch all kernels used in a build env.

kraj, 2018-10-26:
@agherzan has already pointed to right solution and it’s possible that there are more than one scripts in the kernel now which needs to change to use awk from /use/bin it will be good to hunt them all down and prepare a kernel patch 

Doing it via recipe is a workaround which we can have as a workaround

agherzan, 2019-06-14:
@manszigher is this is this case with the current kernel versions?

pbrkr, 2019-11-14:
This is fixed on recent kernels: https://github.com/torvalds/linux/commit/8af11c1cdd8fa08217e702b57cf96e9030db52b2

closed 2019-11-14 by pbrkr