agherzan/meta-raspberrypi #190: NameError: name 'base_conditional' is not defined

diegorusso, 2018-02-01:
When 

**Description**
meta-raspberrypi layer uses old functions from openembedded-core layer.
The author removed them with commit

https://github.com/openembedded/openembedded-core/commit/0391fcad9103abca0796a068f957d0df63ab4776

but base_conditional is still used by meta-raspberrypi layer

https://github.com/agherzan/meta-raspberrypi/blob/master/recipes-kernel/linux/linux-raspberrypi.inc#L25

This raises an error when building:
```
ERROR: ExpansionError during parsing /work/machine-raspberrypi3/manifest/build/conf/../../layers/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.9.bb                                                                                                                                        | ETA:  0:01:00
Traceback (most recent call last):
bb.data_smart.ExpansionError: Failure expanding variable CMDLINE, expression was dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait ${@base_conditional("ENABLE_KGDB", "1", "kgdboc=serial0,115200", "", d)}  ${@base_conditional("DISABLE_RPI_BOOT_LOGO", "1", "logo.nologo", "", d)}  which triggered exception NameError: name 'base_conditional' is not defined
```

gizero, 2018-02-01:
#188 is already there to resolve this

agherzan, 2018-02-01:
#188 was merged. Closing this.

closed 2018-02-01 by agherzan