agherzan/meta-raspberrypi #195: pypi.bbclass removed in latest HEAD of meta-openembedded

erkki, 2018-02-09:
ERROR: ParseError at /workdir/poky/meta-raspberrypi/recipes-devtools/python/rpi-gpio_0.6.3.bb:8: Could not inherit file classes/pypi.bbclass

commit 73dfaca93ba68f63872851adfb0665b82c7078f4
Author: Derek Straka <derek@asterius.io>
Date:   Wed Jan 3 11:31:58 2018 -0500

    pypi: remove class that was added to oe-core
    
    Signed-off-by: Derek Straka <derek@asterius.io>
    Signed-off-by: Armin Kuster <akuster808@gmail.com>

ionutneicu, 2018-02-21:
You need to checkout specific git branch for the following repositories: meta-openembedded, poky and meta-raspberrpi. Use the same release branch for all, don't work on master, i.e. rocko.

https://wiki.yoctoproject.org/wiki/Releases

betafive, 2018-02-26:
pypi.bbclass is now in openembedded-core. Closing this. Please re-open if you still have issues.

closed 2018-02-26 by betafive