agherzan/meta-raspberrypi #198: Issues with console and U-Boot functionality - Rocko branch

mfrydrysiak, 2018-02-10:
**Description**

Hi,
I'm facing an issue with my RPi3. I use the Rocko branches on meta-raspberrypi, meta-oe and poky, the conf/local.conf file contains:
```
ENABLE_UART = "1"
KERNEL_IMAGETYPE = "uImage"
PREFERRED_PROVIDER_virtual/bootloader ?= "u-boot"
UBOOT_MACHINE_raspberrypi3 ?= "rpi_3_32b_defconfig"
RPI_USE_U_BOOT = "1"
MACHINE = "raspberrypi3"
```
I built the image with `bitbake rpi-hwup-image`.

After connecting the RPi3 through the UART/USB FTDI converter I do not get any prints on the console (checked also the baudrate = 115200). If I boot the board with the HDMI cable connected I can see the U-Boot prompt, but none of the bootloader commands work (such as pr, bdinfo, setenv, etc.).

**Steps to reproduce the issue:**
1. Checkout all branches of the following meta-layers to Rocko:
```
/yocto/poky/
/yocto/meta-raspberrypi
/yocto/meta-openembedded
```
2. Append to `conf/local.conf` file settings as listed above
3. Run `bitbake rpi-hwup-image`
4. dd the sd card with the rpi-image image
5. check proper settings in the `config.txt` file on the sd card partition
6. try to boot the RPi3 with the FTDI converter connected 

**Describe the results you received:**
- no output on the console (I use `picocom -b 115200 /dev/ttyUSB0`)
- with the HDMI cable connected, none of the U-Boot commands runs

**Describe the results you expected:**
- at least some prints on the console from the serial connection

**Additional information you deem important (e.g. issue happens only occasionally):**
It would neglect a hardware problem since this does not happen if I build the RPi3 image with buildroot.
**Additional details (revisions used, host distro, etc.):**
Rocko branch

betafive, 2018-02-12:
If you build without `ENABLE_UART = "1"`, can you interact with u-boot via USB keyboard and HDMI monitor? If so that will rule out the most basic problems.

When you say "none of the U-Boot commands runs", are you entering commands via USB keyboard or via the serial console? I want to make sure I understand you correctly.

mfrydrysiak, 2018-02-16:
> If you build without `ENABLE_UART = "1"`, can you interact with u-boot via USB keyboard and HDMI monitor?

Yes, I can interact (although I get the `timeout on interrupt endpoint` warning, but this is U-Boot case for keyboard scanning).

I tried to add manually the UART settings to the `config.txt` file, such as:
`enable_uart=1`
`dtoverlay=pi3-disable-bt`
`dtoverlay=pi3-miniuart-bt`
but none of them work. 

I have to mention that this time I run a clean build without `ENABLE_UART = "1"`, just to be sure that everything is built from scratch.

kraj, 2018-04-16:
@mfrydrysiak do we still need to keep this open ?

agherzan, 2018-04-23:
I'll close this issue for now. Feel free to comment if this still needs attention.

closed 2018-04-23 by agherzan