agherzan/meta-raspberrypi #199: Clean-up of images specific provided by meta-raspberrypi

mirzak, 2018-02-12:
This is a issue based on discussion started in #184 (which is now a closed issue), and to not lose track of the suggestions/discussion I will create a new issue.

To try and summarize the discussion:

There is some confusing today, what the intention of the different images provided by meta-raspberrypi actually is and there is no documentation on them. The current state is:

- `rpi-hwup-image` - Basically equivalent of `core-image-minimal` with kernel-modules installed
- `rpi-basic-image` - Same as above with "splash" and "ssh" enabled
- `rpi-test-image` - Same as above with some additional packages installed suitable for different hw tests (`packagegroup-rpi-test`)

The confusing with above is that NONE actually install all the necessary software/firmware to utilize all the hardware components on the RPi boards. Most confusing is that the WiFi firmware does not get installed using above images.

My suggestions is instead and try and utilize the "default" images provided by Yocto/OE-core which already have a nice layering to them (and documented already in the Yocto docs).

- `core-image-minimal` - "A small image just capable of allowing a device to boot."
- `core-image-base` - "A console-only image that fully supports the target device hardware."

And by settings up `MACHINE_EXTRA_RRECOMMENDS` and `MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS` the correct packages will be installed in above images. This is basically already done today.

This means that we remove the `rpi-hwup-image` and `rpi-basic-image` and change the `rpi-test-image` to inherit from `core-image`, with maybe some additional configuration.

Feedback is welcome what the best approach is.

betafive, 2018-02-14:
I agree with this, meta-raspberrypi should follow the conventions in
oe-core unless we have good reasons not to.

I would keep the old image names around as aliases with deprecation
warnings for at least one release cycle (~ 6 months) so that people can
upgrade safely. So rpi-hwup-image would become an alias for
core-image-minimal and rpi-basic-image would become an alias for
core-image-minimal. I think we should keep rpi-test-image permanently as
it's useful.

-- 
Paul Barker
Togán Labs Ltd

agherzan, 2018-02-14:
I agree with this. I would keep a test image though because it drags all the components we currently have in the layer and it's useful to test a build of all the things we provide (as Paul said).

hhromic, 2018-02-15:
As one of the initiators of the whole discussion, I also agree with this. We totally should be using the standard Yocto images. This in turn makes the whole idea/purpose of Yocto of seamlessly changing BSP layers and then it "just works".
I agree the particular test image for this layer should be kept for the purpose of testing this layer specifically.

I also want to encourage the documentation of this and other features. The best way to stop users being confused (like I was myself at the beginning) is to clearly state how things should work and the best practices. I can help with this as well if you want.

betafive, 2018-02-26:
I've looked into this today and I can't see any way to issue deprecation warnings. What I can do though is make the current `rpi-hwup-image` and `rpi-basic-image` recipes error out with a message telling the user which image recipe to use instead. Then after a release cycle or two we can drop these completely.

I'm going to submit a pull request implementing that as soon as I'm done testing.

agherzan, 2018-02-26:
@betafive Can't we have a warning in do_image for the rpi images? Basically if you are building for example rpi-hw-up-image, warn that next release will have these images removed.

betafive, 2018-02-26:
@agherzan, Actually, yea, that probably will work. I'll give that a go.

betafive, 2018-02-26:
I've added a `bb.warn()` call in a `do_image_prepend()` function. Looks good:

    NOTE: Executing SetScene Tasks
    NOTE: Executing RunQueue Tasks
    WARNING: rpi-hwup-image-1.0-r0 do_image: The image 'rpi-hwup-image' is deprecated, please use 'core-image-minimal' instead
    NOTE: Tasks Summary: Attempted 1574 tasks of which 1546 didn't need to be rerun and all succeeded.
    
    Summary: There was 1 WARNING message shown.

Will submit a pull request once I've finished testing the rest of my changes (kernel & firmware upgrades including a 4.14.y kernel).

agherzan, 2018-02-26:
Exactly. Looks great!

betafive, 2018-02-26:
It's not printed when the image is picked up from sstate but I think we can let that slide.

agherzan, 2018-02-26:
Agreed.

gizero, 2018-02-27:
LGTM. With a similar approach, meta-freescale is warning about future deprecation of their non-wic based image creation classes.

hhromic, 2018-02-27:
great @betafive, exactly what I had in mind!
we should definitively also pay attention to the **documentation**. If someone gets confused by a dissapearing `rpi-hwup-image` or `rpi-basic-image`, then going to the documentation should point them to the right direction. This further makes the sstate "problem" not a problem really.
Thanks for the work!

betafive, 2018-02-27:
@hhromic, you're right. I find coding and documentation writing to need two very different states of mind so I usually do them separately. But lets keep this issue open until the docs are done. Should hopefully be later this week.

hhromic, 2018-02-27:
I can happily help too with that. I guess the relevant files are the ones located under `docs`?
Let me know, however I can work on this only from two weeks from now.
I also want to propose/discuss an improvement for the handling of the options in `config.txt`. This also would require documentation updates. I will open another issue ticket for that later.

pabigot, 2018-03-07:
I second @hhromic's comment on documentation.  It was disconcerting to follow the Quick Start instructions in README to bitbake rpi-hwup-image, only to get a warning that the recipe was deprecated.

agherzan, 2018-03-23:
Docs updated. Closing this issue.

closed 2018-03-23 by agherzan