agherzan/meta-raspberrypi #200: wiringpi compilation issue

BelalElkady, 2018-02-26:
I included wiringPi library in my layer's code This is my C code and I got this issue 

cannot find -lwiringPi
////////////////////////////////////////////////////////
#include <stdio.h>
#include <wiringPi.h>

#define LED 0
#define    OUTPUT 1
extern void pinMode(int pin, int mode) ;
extern int  wiringPiSetup (void);
extern void digitalWrite (int pin, int value) ;
extern void delay (unsigned int howLong) ;

int main(int argc, char **argv)
{
     printf ("Raspberry Pi - Gertboard Blink\n") ;
     wiringPiSetup () ;
     pinMode (LED, OUTPUT) ;
     while(1)
     {
        digitalWrite (LED, 1) ;     // On
        delay (500) ;               // mS
        digitalWrite (LED, 0) ;     // Off
        delay (500) ;
     }
    return 0;
}
////////////////////////////////////////////////////////

And this is the line I added to the example recipe to include compiling wiring pi with this C file
///////////////////////////////////////////////////////
do_compile() {
         ${CC} ${LDFLAGS} helloworld.c -o helloworld -lwiringPi
}
//////////////////////////////////////////////////////

I also added the wiringPi recipe in the rpi-hwup-image.bb as follow
//////////////////////////////////////////////////
# Include modules in rootfs
IMAGE_INSTALL += " \
    kernel-modules \
    example \
    wiringpi \
    "

betafive, 2018-02-26:
You should add the dependencies for your C program to the recipe which builds it, instead of to the IMAGE_INSTALL list. That should ensure that the headers and libraries are available in your do_compile() function.

So in your recipe make sure you have a line like this:

    DEPENDS += "wiringpi"

agherzan, 2018-02-26:
@BelalElkady please user markdown in the description. I updated it this time. And do try Paul's suggestion.

BelalElkady, 2018-02-26:
thank you for your support , the issue is solved

closed 2018-02-26 by agherzan