agherzan/meta-raspberrypi #204: raspberrypi-zero-w: how to enable camera module

DengkeDu, 2018-03-07:
root@raspberrypi0-wifi:~# raspivid -t 0
mmal: mmal_vc_component_create: failed to create component 'vc.ril.camera' (1:ENOMEM)
mmal: mmal_component_create_core: could not create component 'vc.ril.camera' (1)
mmal: Failed to create camera component
mmal: main: Failed to create camera component
mmal: Camera is not enabled in this build. Try running "sudo raspi-config" and ensure that "camera" has been enabled

DengkeDu, 2018-03-07:
> root@raspberrypi0-wifi:~# vcgencmd get_camera
> supported=0 detected=0

DengkeDu, 2018-03-08:
The package rpi-config was null

DengkeDu, 2018-03-08:
Every time I need mount the boot partion, and modify the config.txt.

betafive, 2018-03-08:
Have you tried building with `VIDEO_CAMERA = "1"` set in your local.conf file? Looking at the documentation that should be the setting you need.

agherzan, 2018-03-13:
@DengkeDu Did you try what Paul recommended above? I'd like to close this issue.

DengkeDu, 2018-03-14:
Yes, it works!

closed 2018-03-14 by DengkeDu