agherzan/meta-raspberrypi #208: libbrcmEGL.so undefined symbols with Wayland

tuupola, 2018-03-11:
**Description**

I am trying to use a binary which has been precompiled using Raspbian Stretch. Running it fails with:

```
symbol lookup error: /usr/lib/libbrcmEGL.so: undefined symbol: wl_callback_interface
````

Looking a the library itself it does seem to be missing bunch of symbols, including the Wayland ones. Full output in a [separate gist](https://gist.github.com/tuupola/5ad7039503ae8b5b0b0f1c8d342b8428).

```
# nm -Du  /usr/lib/libbrcmEGL.so | grep wl_
         U wl_buffer_interface
         U wl_callback_interface
         U wl_display_create_queue
         U wl_display_dispatch_pending
        ...
```

I have tried building `core-image-weston`.

```
IMAGE_INSTALL_append = " libevent libpng ffmpeg freetype userland"
```
```
$ bitbake core-image-weston
```

I also have tried building `rpi-basic-image` with `CORE_IMAGE_EXTRA_INSTALL` options in `local.conf`

```
IMAGE_INSTALL_append = " libevent libpng ffmpeg freetype userland"
CORE_IMAGE_EXTRA_INSTALL += "wayland weston"
```
```
$ bitbake rpi-basic-image
```

I have a feeling that I either totally misunderstand something or there is a build setting I am missing.

How can I have the Wayland symbols be included in userland libraries?

**Additional details (revisions used, host distro, etc.):**

```
Build Configuration:
BB_VERSION           = "1.36.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.8"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.4.2"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky            = "rocko:5ba69a97ab5faa8f3866aaeab1d6eaa3cb8149ed"
meta-oe
meta-multimedia
meta-networking
meta-python          = "rocko:a65c1acb1822966c3553de9fc98d8bb6be705c4e"
meta-raspberrypi     = "rocko:b58ccfc696e4fa6b680ae5ca79960312676d7af9"
```

tuupola, 2018-03-11:
Maybe related to https://github.com/agherzan/meta-raspberrypi/pull/202/commits/790e100a90f288e9f8b98fb9ff9bfddd7ad297af?

agherzan, 2018-03-13:
@kraj would be the guru for this.

kraj, 2018-03-14:
@agherzan  merge https://github.com/agherzan/meta-raspberrypi/pull/202
and its fixed in that pull request, it has been pending for over a week without any response.

tuupola, 2018-03-16:
@kraj I tried compiling using #202 but still have the same problem. I am pretty sure I am missing either some DISTRO_FEATURES or IMAGE_INSTALL setting. Could you share yours which you used for testing? It would nudge me to right direction.

tuupola, 2018-03-16:
Also found this:

> In wayland enabled build libbrcmEGL lib is using wayland functions, but during 
the build it is not linking with wayland libraries. Hence if try to link 
libbrcmEGL, will get undefined symbols.

https://www.mail-archive.com/yocto@yoctoproject.org/msg38949.html

tuupola, 2018-03-22:
I still have some undefined symbols but after #202 was merged the original problem seems to be gone. Thanks!

closed 2018-03-22 by tuupola

agherzan, 2018-03-22:
Glad to hear that.