agherzan/meta-raspberrypi #220: Clarify instructions on initramfs support?

cveilleux, 2018-04-10:
**Description**

The documentation at http://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#image-with-initramfs is insufficient to get initramfs working.

**Steps to reproduce the issue:**
1. Follow the instructions at http://meta-raspberrypi.readthedocs.io/en/latest/extra-build-config.html#image-with-initramfs

I tried core-image-tiny-initramfs, core-image-minimal, rpi-hwup-image. They all failed.

2. Attempt to build an image: `bitbake core-image-full-cmdline`

**Describe the results you received:**

```
ERROR: Nothing PROVIDES 'core-image-tiny-initramfs'
core-image-tiny-initramfs was skipped: incompatible with host arm-poky-linux-gnueabi (not in COMPATIBLE_HOST)
```

**Describe the results you expected:**

I think I somehow need to provide my own, rpi compatible, image that would be suitable for booting. I have no idea how to do that.

kraj, 2018-04-11:
regular images are not suitable for bundling into initramfs so you have to define a initramfs image

Something like this ```core-image-minimal-initramfs.bb```

```
require meta/recipes-core/images/core-image-minimal.bb

IMAGE_FSTYPES_rpi = " ${INITRAMFS_FSTYPES}"
```

cveilleux, 2018-04-11:
Thanks for the reply. I will do some tests and will clarify the docs if I succeed. Please leave issue open.

agherzan, 2018-05-09:
@cveilleux Any updates on this?

agherzan, 2018-07-26:
Closing this due to inactivity. I've tested initramfs images and everything seems to work as expected.

closed 2018-07-26 by agherzan