agherzan/meta-raspberrypi #228: Building core-image-x11 for raspberrypi3-64 on master currently failing

betafive, 2018-04-21:
In `conf/machine/include/rpi-base.inc` we define `XSERVER` to contain `xserver-xorg-extension-glx` if `MACHINE_FEATURES` contains `vc4graphics`. This is picked up as the dependencies for `packagegroup-core-x11-xserver`.

However, `xserver-xorg-extension-glx` is only built if the `PACKAGECONFIG` for `xserver-xorg` contains `glx`. In our `xserver-xorg_%.bbappend` file we gate this on the presence of `opengl` in `DISTRO_FEATURES`.

So by default trying to install `packagegroup-core-x11-xserver` fails:

    pbarker@eng01 ~/projects/oe-nodistro/build $ MACHINE=raspberrypi3-64 bitbake core-image-x11
    Loading cache: 100% |###########################################################################################################################################################| Time: 0:00:00
    Loaded 2950 entries from dependency cache.
    NOTE: Resolving any missing task queue dependencies

    Build Configuration:
    BB_VERSION           = "1.37.0"
    BUILD_SYS            = "x86_64-linux"
    NATIVELSBSTRING      = "ubuntu-16.04"
    TARGET_SYS           = "aarch64-oe-linux-musl"
    MACHINE              = "raspberrypi3-64"
    DISTRO               = "nodistro"
    DISTRO_VERSION       = "nodistro.0"
    TUNE_FEATURES        = "aarch64"
    TARGET_FPU           = ""
    meta                 = "HEAD:1f53140528d79c38d4f3a82cd0a03bd0ddc87275"
    meta-oe
    meta-multimedia
    meta-networking
    meta-python          = "HEAD:f0a664f2b7ac42fa8832ecc016ddf8b6260ee50e"
    meta-raspberrypi     = "master:3175e7f3e2066ef34897013742ab95341c3ebaae"

    Initialising tasks: 100% |######################################################################################################################################################| Time: 0:00:03
    NOTE: Executing SetScene Tasks
    NOTE: Executing RunQueue Tasks
    ERROR: core-image-x11-1.0-r0 do_rootfs: Unable to install packages. Command '/home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/recipe-sysro
    ot-native/usr/bin/opkg --volatile-cache -f /home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/opkg.conf -t /home/pbarker/projects/oe-nodistr
    o/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/temp/ipktemp/ -o /home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/roo
    tfs  --force_postinstall --prefer-arch-to-version   install opkg packagegroup-base-extended packagegroup-core-boot packagegroup-core-x11-base psplash-raspberrypi run-postinsts' returned 1:
    Collected errors:
     * Solver encountered 1 problem(s):
     * Problem 1/1:
     *   - nothing provides xserver-xorg-extension-glx needed by packagegroup-core-x11-xserver-1.0-r40.raspberrypi3_64
     *
     * Solution 1:
     *   - do not ask to install a package providing packagegroup-core-x11-base

     * opkg_finalize_intercepts: Failed to open dir /home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/temp/ipktemp//opkg-AhNgIa/opkg-intercept-
    rgrYhG: No such file or directory.
     * rm_r: Failed to open dir /home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/temp/ipktemp//opkg-AhNgIa/opkg-intercept-rgrYhG: No such file
     or directory.
     * rm_r: Failed to open dir /home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/temp/ipktemp//opkg-AhNgIa: No such file or directory.

    ERROR: core-image-x11-1.0-r0 do_rootfs: Function failed: do_rootfs
    ERROR: Logfile of failure stored in: /home/pbarker/projects/oe-nodistro/build/tmp/work/raspberrypi3_64-oe-linux-musl/core-image-x11/1.0-r0/temp/log.do_rootfs.13345
    ERROR: Task (/home/pbarker/projects/oe-nodistro/openembedded-core/meta/recipes-graphics/images/core-image-x11.bb:do_rootfs) failed with exit code '1'
    NOTE: Tasks Summary: Attempted 5237 tasks of which 4236 didn't need to be rerun and 1 failed.

I've made minimal changes to `local.conf` which shouldn't break things:

    LICENSE_FLAGS_WHITELIST = "commercial"
    TCLIBC = "musl"
    TCLIBCAPPEND = ""
    ROOT_HOME = "/root"
    COPY_LIC_MANIFEST = "1"
    COPY_LIC_DIRS = "1"
    BB_GENERATE_MIRROR_TARBALLS = "1"
    INHERIT += "rm_work"
    RPI_USE_U_BOOT = "1"

Otherwise this is a pretty much out-of-the-box build.

There's two ways we can go to fix this:

1) Add `opengl` to `REQUIRED_DISTRO_FEATURES` for xserver-xorg when `MACHINE_FEATURES` contains `vc4graphics`.

2) Drop `xserver-xorg-extension-glx` from `XSERVER` unless `DISTRO_FEATURES` contains `opengl`.

Both approaches seem to work but as I'm not using xorg extensively I'm probably not the best person to make the decision. Is opengl a requirement for a basic x11 or sato image on raspberrypi3-64?

kraj, 2018-04-22:
We need this extention so option 2 is not good but I am ok with Option 1.

agherzan, 2018-04-23:
Vote for Option 1 too.

betafive, 2018-05-09:
Apologies, have been pretty busy lately. This is still on my list though and I'm aiming to send PRs for master & sumo when I get chance to look at this.

agherzan, 2018-07-26:
Friendly ping on this, @betafive .

betafive, 2018-07-27:
I'd totally forgot about this due to a few changes with work, I'll put it back onto the todo list!

Romaxx, 2018-10-08:
now build (core-image-x11) is successful but x11 not launching (but launching with the same config with MACHINE = "raspberrypi3" instead of MACHINE = "raspberrypi3-64"), in this case, Xorg said 
> (EE) no screens found(EE)  
Some advices ? Thanks

kraj, 2018-10-08:
I fixed it last week to boot on raspberrypi3-64, I wonder if you have those changes.

Romaxx, 2018-10-09:
Yes, I have pulled the last master version to build the image, Xorg doesn't detect the screen
`cat /dev/urandom > /dev/fb0` ok
`FRAMEBUFFER=/dev/fb0 startx` screen not found: command output:

> X.Org X Server 1.20.1
> X Protocol Version 11, Revision 0
> Build Operating System: Linux 4.15.0-36-generic x86_64 
> Current Operating System: Linux raspberrypi3-64 4.14.68 #1 SMP PREEMPT Mon Oct 8 08:49:47 UTC 2018 aarch64
> Kernel command line: 8250.nr_uarts=1 bcm2708_fb.fbwidth=1024 bcm2708_fb.fbheight=768 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
> Build Date: 08 October 2018  10:30:31AM
> Current version of pixman: 0.34.0
> 	Before reporting problems, check http://wiki.x.org
> 	to make sure that you have the latest version.
> Markers: (--) probed, (**) from config file, (==) default setting,
> 	(++) from command line, (!!) notice, (II) informational,
> 	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
> (==) Log file: "/var/log/Xorg.0.log", Time: Mon Oct  8 22:56:04 2018
> (==) Using config file: "/etc/X11/xorg.conf"
> (==) Using system config directory "/usr/share/X11/xorg.conf.d"
> (EE) 
> Fatal server error:
> (EE) no screens found(EE) 

/var/log/Xorg.0.log

> [   158.135] (==) ModulePath set to "/usr/lib/xorg/modules"                      
> [   158.135] (II) The server relies on udev to provide the list of input devices.
>         If no devices become available, reconfigure udev or disable AutoAddDevices.
> [   158.135] (II) Loader magic: 0x5579dbac28                                       
> [   158.135] (II) Module ABI versions:                                             
> [   158.135]    X.Org ANSI C Emulation: 0.4                                        
> [   158.135]    X.Org Video Driver: 24.0                                           
> [   158.135]    X.Org XInput driver : 24.1                                         
> [   158.135]    X.Org Server Extension : 10.0                                      
> [   158.135] (II) no primary bus or device found                                   
> [   158.135] (II) LoadModule: "glx"                                                
> [   158.136] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so               
> [   158.150] (II) Module glx: vendor="X.Org Foundation"                            
> [   158.150]    compiled for 1.20.1, module version = 1.0.0                        
> [   158.150]    ABI class: X.Org Server Extension, version 10.0                    
> [   158.150] (==) Matched modesetting as autoconfigured driver 0                   
> [   158.150] (==) Matched fbdev as autoconfigured driver 1                         
> [   158.150] (==) Assigned the driver to the xf86ConfigLayout                      
> [   158.150] (II) LoadModule: "modesetting"                                        
> [   158.151] (II) Loading /usr/lib/xorg/modules/drivers/modesetting_drv.so         
> [   158.151] (II) Module modesetting: vendor="X.Org Foundation"                    
> [   158.151]    compiled for 1.20.1, module version = 1.20.1                       
> [   158.151]    Module class: X.Org Video Driver                                   
> [   158.151]    ABI class: X.Org Video Driver, version 24.0                        
> [   158.151] (II) LoadModule: "fbdev"                                              
> [   158.152] (WW) Warning, couldn't open module fbdev                              
> [   158.152] (EE) Failed to load module "fbdev" (module does not exist, 0)         
> [   158.152] (II) modesetting: Driver for Modesetting Kernel Drivers: kms          
> [   158.152] (--) using VT number 3                                                
>                                                                                    
> [   158.162] (WW) Falling back to old probe method for modesetting        
> [   158.162] (EE) open /dev/dri/card0: No such file or directory          
> [   158.162] (WW) Falling back to old probe method for modesetting        
> [   158.162] (EE) open /dev/dri/card0: No such file or directory          
> [   158.162] (EE) No devices detected.                                    
> [   158.162] (EE)                                                         
> Fatal server error:                                                       
> [   158.162] (EE) no screens found(EE)

Romaxx, 2018-10-09:
DISTRO_FEATURES_append = " opengl"
MACHINE_FEATURES_append=" vc4graphics"
I want to use core-image-x11 for meta-browser/chromium-x11 and 64bit version for mongodb (V4 is 64bits only)
using meta-raspi-light instead of meta-raspberrypi works for 64bit version (I have tried MACHINE_FEATURES_remove=" wayland" with meta-raspberrypi but same problem).

kraj, 2018-10-12:
I built core-image-x11 on latest master and it booted fine with vc4graphics on raspberrypi3-64, if there are patches to build it with userland graphics, that will be awesome if you find them and post.

kraj, 2018-11-28:
@Romaxx this works fine for me on rpi3, Can you try it with latest master/thud

Romaxx, 2018-11-28:
Hi, now it's working :)

kraj, 2018-11-28:
thanks for confirming, I am closing this issue, if something pops up. Feel free to reopen

closed 2018-11-28 by kraj