agherzan/meta-raspberrypi #230: Dropbear - i can not ssh the device

sherifomran, 2018-04-28:
**Description**

I installed a new image and i am trying to use it from another computer due to a missing keyboard, i tried to ssh the device but i can not connect to it. It seems there are some ports closed.

**Steps to reproduce the issue:**
bitbake rpi-basic-image

agherzan, 2018-05-09:
@sherifomran Did you try what was recommended above?

Manyyack, 2018-07-12:
ssh doesn't work for me when I do 'core-image-base' but working on rpi-basic-image

agherzan, 2018-07-26:
@Manyyack `rpi-basic-image` is deprecated. You can define your own image and include `ssh-server-dropbear`. Indeed `core-image-base` doesn't do that and it is not supposed to either. Yocto provides all the tools needed to define your own image based on your needs.

Closing this as it looks like a configuration issue. We will reopen if proven otherwise.

closed 2018-07-26 by agherzan