agherzan/meta-raspberrypi #236: incorrect board address in RPi3B+ bluetooth

pabigot, 2018-05-06:
While adding the firmware makes bluetooth work on RPi3B+, the Yocto distribution doesn't get the right board address: two distinct boards both come up as 43:45:c0:00:1f:ac.  When I boot one of the devices with raspbian I do get a proper address with a Raspberry Pi OUI.

hciattach, if given the correct board address on the command line, will write it.  So I'm guessing that something is supposed to be pulling the board address out of thin air and programming it as the default but is not doing so in meta-raspberrypi.  I have no idea what that something is.  Differences between the bluez 5.48 Yocto's using and the 5.43 used by raspbian in hciattach do not seem relevant, though AFAICT raspbian is using none of the patches that meta-raspberrypi has.

pabigot, 2018-05-06:
Appears this is raspberrypi/firmware#860 and requires a modification to the hciattach command to calculate the device-specific bluetooth address from the device serial number, done on raspbian by a special /usr/bin/btuart script that runs hciattach.  I'll work on that.

pabigot, 2018-05-06:
Some notes for posterity:

Device addresses are generally obtained from a one-time-programmable area that's being documented (raspberrypi/firmware#974).

A Raspberry Pi 3B with serial number 00000000685e14b0 (`cat  /proc/cpuinfo | grep Serial`) has these addresses:
* `eth  b8:27:eb:5e:14:b0` : low 24 bits match the serial number
* `wifi b8:27:eb:0b:41:e5` : low 24 bits are nybble-reversed serial number
* `bt   b8:27:eb:f4:be:1a` : low 24-bits are nybble-reversed serial number bitwise inverted

A Raspberry Pi 3B+ serial number 000000003250ab5a booted under Raspbian has
* `eth  b8:27:eb:50:ab:5a` : low 24 bits match the serial number
* `wifi b8:27:eb:05:fe:0f` : low 24 bits are serial number XORd with 0x555555
* `bt   b8:27:eb:fa:01:f0` : low 24 bits are serial number XORd with 0xAAAAAA

agherzan, 2018-05-06:
Very nice. It would be nice to have this documented as well.

pabigot, 2018-05-06:
Dunno if they plan to put it in upstream documentation.

Just as a heads-up: between this and #224 I'm going to have to do some refactoring to clean up the licensing.  The firmware images are under a Cypress license, while the script that sets the device address is BSD-3-Clause.  I don't think either of those should be bundled into bluez which is GPLv2+ & LGPLv2.1+.

The simplest way is probably going to be to move the firmware to a new raspbian-bluez-firmware package, and to put the systemd script support that uses [raspbian btuart](https://github.com/RPi-Distro/pi-bluetooth) into its own package too.  It'll take a couple days to get around to this.

closed 2018-05-09 by agherzan

pabigot, 2018-05-14:
@agherzan huh?  The examples I gave above include ones from a 3B+, albeit in 32-bit mode.  However, meta-raspberrypi is currently not loading the correct device tree for aarch64, which I'm looking at for #251; the lan78xx driver will take what the device tree says for the MAC address.

agherzan, 2018-05-14:
Something is strange on B+.

pabigot, 2018-05-14:
See #254.  I've confirmed that the correct eth0 MAC address (i.e., consistent with the algorithm above) is read from OTP once the correct device tree is loaded.

agherzan, 2018-05-14:
@pabigot makes sense.

commit 53af636f (2018-05-09): pi-bluetooth: create to provide standard raspbian script for bluetooth setup

Unique bluetooth device addresses are not assigned correctly with some
firmware, resulting in a fixed default address.  Use the same utility as
raspbian does to initialize the device.

Closes #236
commit 6aa54e92 (2018-05-11): pi-bluetooth: create to provide standard raspbian script for bluetooth setup

Unique bluetooth device addresses are not assigned correctly with some
firmware, resulting in a fixed default address.  Use the same utility as
raspbian does to initialize the device.

Closes #236