agherzan/meta-raspberrypi #243: libepoxy build failure with tools-profile in IMAGE_FEATURES

gizero, 2018-05-10:
I observed that images which include `tools-profile` in `IMAGE_FEATURES` systematically fail to build with:

```
ERROR: libepoxy-1.5.0-r0 do_prepare_recipe_sysroot: The file /usr/lib/libwayland-egl.so is installed by both wayland and userland, aborting
ERROR: libepoxy-1.5.0-r0 do_prepare_recipe_sysroot: Function failed: extend_recipe_sysroot
ERROR: Logfile of failure stored in: /home/gizero/work/smartliving/distro/repo-master/build-poky/tmp/work/cortexa7hf-neon-vfpv4-poky-linux-gnueabi/libepoxy/1.5.0-r0/temp/log.do_prepare_recipe_sysroot.28767
ERROR: Task (/home/gizero/work/smartliving/distro/repo-master/build-poky/conf/../../layers/poky/meta/recipes-graphics/libepoxy/libepoxy_1.5.0.bb:do_prepare_recipe_sysroot) failed with exit code '1'
``` 
Happens with custom "debug" image, but it's reproducible with `core-image-minimal` when EXTRA_IMAGE_FEATURES includes `tools-profile`.
Cleaning up relevant recipes doesn't help. Removing `tools-profile` brings things back to normality. 

This happens with the following commits:

```
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta
meta-poky
meta-yocto-bsp       = "HEAD:f4c938c47424424b89cde2269bd92cebc9a6ac1a"
meta-raspberrypi     = "master:53af636f88d13141e70e53bc33d571a207122b03"
```

I haven't noticed it in the past with the same images. Any recent change either on `poky` or in this repo that come to mind? I'm happy to further investigate on this, but I probably need a hint what to look for...

hhromic, 2018-05-10:
I can confirm this also started happening to me when building `meta-wpe` in the `master` branch for all components.

hhromic, 2018-05-10:
I noticed that Wayland was updated recently in Poky, referencing `libwayland-egl` being moved to `libwayland` and some recipes (such as `userland`?) needing updates:

```
commit 8097bf7012a0d97d3b8b48eceb5dc20cc4ae7ca9
Author: Denys Dmytriyenko <denys@ti.com>
Date:   Tue Apr 10 15:06:09 2018 -0400

    wayland: upgrade to 1.15.0

    Official announcement:
    https://lists.freedesktop.org/archives/wayland-devel/2018-April/037767.html

    | libwayland-egl is now part of libwayland, and will presumably be removed
    | from mesa in the not too distant future.

    Update mesa recipe by removing corresponding libwayland-egl entries.

    (From OE-Core rev: 6e5952fcfc13ff4b63c9376bd41a1dbba957f425)

    Signed-off-by: Denys Dmytriyenko <denys@ti.com>
    Signed-off-by: Ross Burton <ross.burton@intel.com>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>
```

hhromic, 2018-05-10:
This is how `mesa` is dealing with the change in `poky/meta/recipes-graphics/mesa/mesa.inc`:

```
do_install_append () {
    # Drivers never need libtool .la files
    rm -f ${D}${libdir}/dri/*.la
    rm -f ${D}${libdir}/egl/*.la
    rm -f ${D}${libdir}/gallium-pipe/*.la
    rm -f ${D}${libdir}/gbm/*.la

    # it was packaged in libdricore9.1.3-1 and preventing upgrades when debian.bbclass was used
    rm -f ${D}${sysconfdir}/drirc
    chrpath --delete ${D}${libdir}/dri/*_dri.so || true

    # libwayland-egl has been moved to wayland 1.15+
    rm -f ${D}${libdir}/libwayland-egl.so*
    rm -f ${D}${libdir}/pkgconfig/wayland-egl.pc
}
```

hhromic, 2018-05-10:
I think I found a solution based on the above findings: remove `libwayland-egl.so` as done for the `mesa` recipe but in the `userland` recipe.
I'm testing a build now and will send a PR shortly.

hhromic, 2018-05-10:
I'm actually wondering if the `userland` version of `libwayland-egl.so` should be installed and the wayland one discarded. Let's wait for the seniors to give an opinion. @kraj , @betafive , @agherzan , @pabigot  ?

kraj, 2018-05-10:
I would prefer the one coming from wayland, however I dont have high hopes weather this will work. So please do two things, 

1. Update userland to latest master
2. Test with libwayland-egl coming from wayland itself

hhromic, 2018-05-11:
@kraj , will do. As for testing, I'm not sure how to properly test this.  I can test the build succeeds (which it does when removing the libs from the install phase), but I don't know how to test the library itself shipped with wayland.
For example I don't know if the library has a dependency to VC4 graphics from userland. Why it is provided by userland (and mesa)? The version from wayland is agnostic and universal?. I can investigate more but I need some pointers as I am completely foreign to wayland/egl/userland.
If you can give me some ideas on how to test this properly, I can work them out and help.

hhromic, 2018-05-11:
Also, it looks like the wayland library is created in userland from a set of patches in the recipe. If we end up removing the library from the install phase, wouldn't make more sense to not build it in the first place?

gizero, 2018-05-11:
@hhromic any chance for you to share your work-in-progress with this one? This is now getting in my way while looking at #246

hhromic, 2018-05-11:
hi @gizero , to make the build succeed, apply this patch to `meta-raspberrypi`:
```diff
index 7f2dca6..f8c64e6 100644
--- a/recipes-graphics/userland/userland_git.bb
+++ b/recipes-graphics/userland/userland_git.bb
@@ -14,11 +14,11 @@ COMPATIBLE_MACHINE = "^rpi$"

 SRCBRANCH = "master"
 SRCFORK = "raspberrypi"
-SRCREV = "11389772c79685442e0ab8aa9be8ad0e32703f68"
+SRCREV = "a343dcad1dae4e93f4bfb99496697e207f91027e"

 # Use the date of the above commit as the package version. Update this when
 # SRCREV is changed.
-PV = "20180219"
+PV = "20180329"

 SRC_URI = "\
     git://github.com/${SRCFORK}/userland.git;protocol=git;branch=${SRCBRANCH} \
@@ -64,6 +64,10 @@ do_install_append () {
                sed -i 's/include "vcos_futex_mutex.h"/include "pthreads\/vcos_futex_mutex.h"/g' ${f}
                sed -i 's/include "vcos_platform_types.h"/include "pthreads\/vcos_platform_types.h"/g' ${f}
        done
+
+       # libwayland-egl has been moved to wayland 1.15+
+       rm -f ${D}${libdir}/libwayland-egl.so*
+       rm -f ${D}${libdir}/pkgconfig/wayland-egl.pc
 }
```

This will remove `libwayland-egl` installed from the `userland` thus stopping the conflict with `wayland`, but will use the latter's version of the library. With this, the build succeeds.

However I haven't tried any application using wayland+EGL yet so I don't know if this solution is appropiate. If you can test and confirm that doing this keeps EGL-accelerated wayland, I will work-out the final PR to submit.
I'm also thinking on dropping patches that create `libwayland-egl` in `userland` in the first place, but first I would like to confirm the version shipped from `wayland` is appropiate for the RPI.
Thanks!

hhromic, 2018-05-11:
Sorry I just revised the patch, use the edited version in the last comment.

agherzan, 2018-05-11:
It needs testing for sure, but most probably we will need to use the userland provided library for wayland-egl. I don't have any testing setup now so I think it only comes down to what testing reveals. To unblock this I would just go ahead for now and remove the one in wayland unless somebody gets on some testing.

hhromic, 2018-05-11:
I just realised there are testing example applications in `userland` regarding wayland-egl.
However as you suggest, to be sure, for now I will make a PR that removes the `wayland` provided library instead of the `userland` one.
After @gizero and myself test this better, I can provide another PR with a final solution.

agherzan, 2018-05-11:
Could we also use yocto's update-alternatives for this?

hhromic, 2018-05-11:
Ah the alternatives system.. indeed that would be a very elegant solution, specially for multi-board builds.
However I have to learn how to implement it here as I have no experience with it in Yocto.
Feel free to merge the current "crude" PR for now until I (or someone else?) figure out using alternatives for this.

gizero, 2018-05-11:
>@gizero , can you rename this issue to reflect that it's a conflict between wayland and userland providing libwayland-egl ?
This way we don't loose track of the actual problem. Thanks!

title updated!

PR #247 (2018-05-14): wayland: do not install libwayland-egl from wayland
Wayland was updated recently in Poky, and now includes `libwayland-egl`:

    commit 8097bf7012a0d97d3b8b48eceb5dc20cc4ae7ca9
    Author: Denys Dmytriyenko <denys@ti.com>
    Date:   Tue Apr 10 15:06:09 2018 -0400

        wayland: upgrade to 1.15.0

        Official announcement:
        https://lists.freedesktop.org/archives/wayland-devel/2018-April/037767.html

        | libwayland-egl is now part of libwayland, and will presumably be removed
        | from mesa in the not too distant future.

        Update mesa recipe by removing corresp [... cut]
closed 2018-05-14 by agherzan