agherzan/meta-raspberrypi #244: building SDKs fail after intruducing linux-firmware-raspbian

gizero, 2018-05-10:
After recent changes introducing linux-firmware-raspbian recipe, building eSDK fails with:

```
ERROR: core-image-base-dlms-1.0-r0 do_sdk_depends: The file /lib/firmware/LICENCE.cypress is installed by both linux-firmware and bluez-firmware-raspbian, aborting
ERROR: core-image-base-dlms-1.0-r0 do_sdk_depends: Function failed: extend_recipe_sysroot
ERROR: Logfile of failure stored in: /home/gizero/work/smartliving/distro/repo-master/build-poky/tmp/work/raspberrypi3-poky-linux-gnueabi/core-image-base-dlms/1.0-r0/temp/log.do_sdk_depends.1343
ERROR: Task (/home/gizero/work/smartliving/distro/repo-master/build-poky/conf/../../layers/meta-bsl/recipes-core/images/core-image-base-dlms.bb:do_sdk_depends) failed with exit code '1'
```

For me, this happens when the image depends on packages from both linux-firmware and linux-firmware-raspbian. Since the latter does provide RPi-specific firmware only, I think mixing and matching packages from both recipes is to be considered a valid use case. To be more specific, the offending configuration bit for me comes from adding linux-firmware-ath9k to the image.

Moreover the conflicting LICENCE file is not the same at every build, but I think this is expected, having many of them being installed by both recipes.

gizero, 2018-05-10:
Does declaring `RCONFLICTS` or `RREPLACES` for `linux-firmware-raspbian` packages make any sense... I've never used them in practive.

hhromic, 2018-05-10:
Just to clarify, looks like the problem you report is for the `bluez-firmware-raspbian` package, not the (also recent) `linux-firmware-raspbian`.

pabigot, 2018-05-10:
Interesting.

If you are installing both `linux-firmware` and `linux-firmware-raspbian` then things are probably going to get messy with lots of conflicts, but I can argue it's not my problem to fix.

`bluez-firmware-raspbian` provides `LICENCE.cypress` because `linux-firmware-raspbian` does not (though it almost certainly should, see comments in RPi-Distro/bluez-firmware#1).  I expect `bluez-firmware-raspbian` does need to put the license in a separate package that is RCONFLICTS with something.  That is my problem to fix.

The question I have is where are you finding a `linux-firmware` recipe that installs and packages `LICENCE.cypress`?  The one in poky doesn't seem to do so.

pabigot, 2018-05-10:
I can't reproduce this.  There are is a conflict between `linux-firmware` and `linux-firmware-raspbian` that could be partly addressed by adding this to linux-firmware-raspbian:

    RCONFLICTS_${PN}-bcm43430 += "linux-firmware-bcm43430"
    RREPLACES_${PN}-bcm43430 += "linux-firmware-bcm43430"

but even with that you can't just blindly install `linux-firmware` alongside `linux-firmware-raspbian` as they both provide `/lib/firmware/brcm/brcmfmac43455-sdio.bin` (the patch above deconflicts only at the individual driver level, and maybe allows you to ask for `linux-firmware-bcm43430` and get the one from `linux-firmware-raspbian`).  Installing `bluez-firmware-raspbian`, `linux-firmware-raspbian`, and `linux-firmware-ath9k` doesn't produce a conflict for me.

Without further information I can't do anything about conflicts between `LICENSE.cypress` and other packages.

kraj, 2018-05-10:
another idea would be to copy needed bits from raspbian as patches and overlay them on top of upstream version as needed. This will make packaging issues a bit easier to handle. 

Second approach could be to dump upstream version completely and CONFLICT with it at build time.

pabigot, 2018-05-10:
@kraj I think the second approach is what is being used: `linux-firmware-raspbian` only packages two firmware files, and only one of those is also packaged by `linux-firmware` (though the other is present upstream).  So maybe:

    RREPLACES_${PN}-bcm43430 += "linux-firmware-bcm43430"
    RCONFLICTS_${PN}-bcm43430 += "linux-firmware-bcm43430 linux-firmware"

which I'm testing now.

I think the first approach is the one that was discarded in favor of the separate recipe in #215.

gizero, 2018-05-10:
@hhromic You are right! Sorry for the confusion... I saw the `linux-firmware-raspbian`/`linux-firmware` conflict a few days ago and only took the time today to file this issue. However, while double checking what was the minimal setup to reproduce it, I didn't notice that with a clean build I hit the even newer `linux-firmware`/`bluez-firmware-raspbian` and copy pasted its log instead.
@pabigot Are you building eSDK such as with `bitbake core-image-base-dlms -c populate_sdk_ext`? Strangely enough here is 100% reproducible. Tomorrow I'll give your `RRCONFLICTS`/`RREPLACES` approach a go and report back.

pabigot, 2018-05-10:
I don't use the SDK features and didn't know what you meant by "building eSDK".

I can reproduce it with `-c populate_sdk_ext`.  Changes to the meta-raspberrypi recipes won't affect the behavior because that brings in the entire `linux-firmware` package first even though it's not an image dependency.  This is why `LINUX.cypress` shows up: it only exists in the roll-up package.  I don't know where that dependency is coming from; somewhere deep in the bowels of the sdk support, presumably.

I can't see why the whole package is being installed, but the `RCONFLICTS/RREPLACES` won't work because as best I can tell after #215 you can install any of the individual driver packages but you can't install the whole `linux-firmware`.  @agherzan was that tested?

My best guess is `bluez-firmware-raspbian` needs to be removed and its effect produced through a `bbappends` to `linux-firmware`.  It's probably also necessary to do that for `linux-firmware-raspbian`.

agherzan, 2018-05-11:
@pabigot Yes, it was tested but obviously not in the combination with linux-firmware. The core issue here is that both packages install the same license file. I'm all for reunifying with linux-firmware. The recipe separation seems like being clearer but I see how this can backfire.

commit f3ecec38 (2018-05-14): linux-firmware: restore and merge bluez-firmware

The attempt to Raspbian updated firmware blobs in packages separate from
linux-firmware introduced unresolvable conflicts with the standard
linux-firmware roll-up package.  Revert to using an augmented
linux-firmware recipe that overrides and adds firmware from two Raspbian
repositories that have up-to-date images.

Closes #244
commit 092d42cd (2018-05-14): linux-firmware: restore and merge bluez-firmware

The attempt to Raspbian updated firmware blobs in packages separate from
linux-firmware introduced unresolvable conflicts with the standard
linux-firmware roll-up package.  Revert to using an augmented
linux-firmware recipe that overrides and adds firmware from two Raspbian
repositories that have up-to-date images.

Closes #244
closed 2018-05-14 by agherzan