agherzan/meta-raspberrypi #246: gstreamer: upgraded to 1.14 in Poky

hhromic, 2018-05-11:
Hello,
Poky recently upgraded gstreamer to version `1.14`. The gstreamer recipes in this layer should be migrated as there is only one version now (no more `1.12.x`).
I'm creating this issue as a reminder to do this in a near future.
@kraj seems to have experience with this components, if you tell me how to approach the migration I can help creating the PR.

The commit in Poky:
```
commit 37e1470a9a40e3558c9df8012fd26dc4fe205920
Author: Anuj Mittal <anuj.mittal@intel.com>
Date:   Thu May 3 16:17:47 2018 +0800

    gstreamer1.0: upgrade 1.12.4 -> 1.14.0

    * For changes, please see: https://gstreamer.freedesktop.org/releases/1.14/
    * Merge inc and bb file since we only have one version now.

    (From OE-Core rev: 21229feea228c44f2b3851ddd52d1899931040b2)

    Signed-off-by: Anuj Mittal <anuj.mittal@intel.com>
    Signed-off-by: Ross Burton <ross.burton@intel.com>
    Signed-off-by: Richard Purdie <richard.purdie@linuxfoundation.org>
```

agherzan, 2018-05-11:
@hhromic The patches don't apply anymore?

hhromic, 2018-05-11:
@agherzan , I haven't work on this yet. I plan to take a look during the weekend. I plan to:
1. check which patches aren't relevant anymore
2. rebase patches as necessary
3. verify the builds succeed without warnings

if anybody wants to step-in, it is the right time as I haven't done anything yet here.

gizero, 2018-05-11:
@agherzan @hhromic I can confirm some of the patches don't apply anymore... I'm looking into it right now, but I have no experience with that recipe in particular. I'll post an update later.

agherzan, 2018-05-11:
Ok. I was about to pick it up but if you guys are already on it, it's okay.

gizero, 2018-05-11:
To clarify, two of the patches don't apply anymore. More specifically, they are:

- `0002-fix-decoder-flushing.patch`
- `0005-Don-t-abort-gst_omx_video_dec_set_format-if-there-s-.patch`

Both are not trivial to port, at least for me. Surrounding context has changed quite a bit and I don't feel confident enough with the code to blindly "adapt" them. Neither the original patch files nor commit messages from `meta-raspberrypi` do include enough context about the original issues those patches were supposed to solve. Maybe the original author/commiter @kraj can jump in and help with those.

All other patches still apply with no need for adaptations.

gizero, 2018-05-11:
After a closer look, IMHO `0005-Don-t-abort-gst_omx_video_dec_set_format-if-there-s-.patch` sounds useless to me.

```diff
From 0d2ad639e6158c8023c157e206ef3ff7abdc089c Mon Sep 17 00:00:00 2001
From: =?UTF-8?q?Enrique=20Oca=C3=B1a=20Gonz=C3=A1lez?= <eocanha@igalia.com>
Date: Fri, 4 Dec 2015 18:39:59 +0100
Subject: [PATCH] Don't abort gst_omx_video_dec_set_format() if there's a
 timeout releasing the buffers taken by the egl_render out port

---
 omx/gstomxvideodec.c | 5 ++++-
 1 file changed, 4 insertions(+), 1 deletion(-)

diff --git a/omx/gstomxvideodec.c b/omx/gstomxvideodec.c
index bad6335..c63b972 100644
--- a/omx/gstomxvideodec.c
+++ b/omx/gstomxvideodec.c
@@ -1911,8 +1911,11 @@ gst_omx_video_dec_set_format (GstVideoDecoder * decoder,
               5 * GST_SECOND) != OMX_ErrorNone)
         return FALSE;
       if (gst_omx_port_wait_buffers_released (out_port,
-              1 * GST_SECOND) != OMX_ErrorNone)
+              1 * GST_SECOND) != OMX_ErrorNone) {
+#if !(defined (USE_OMX_TARGET_RPI) && defined (HAVE_GST_GL))
         return FALSE;
+#endif
+      }
       if (gst_omx_port_deallocate_buffers (self->dec_in_port) != OMX_ErrorNone)
         return FALSE;
       if (gst_omx_video_dec_deallocate_output_buffers (self) != OMX_ErrorNone)
```

I can't find `USE_OMX_TARGET_RPI` defined anywhere in this layer. Then, the conditional this patch is adding always resolve to true, leaving the original code unchanged. Am I missing something? Otherwise I'd suggest dropping this patch altogether.