agherzan/meta-raspberrypi #251: RPI 3B+ network not working

GreSuh, 2018-05-13:
Build image core-image-full-cmdline for Raspberry3-64.
Image starts without errors but dhcp doesn't work. 

With tcpdump enabled and eth0 in promiscous mode DHCP works.
Seems linke some initialization of eth0 is missing.

**Steps to reproduce the issue:**
Build image  core-image-full-cmdline
openembedded-core
meta-poky
meta-oe
meta-python
meta-perl
meta-networking
meta-security
meta-raspberrypi_priv https://github.com/GreSuh/meta-raspberrypi

but same problem with latest https://git.yoctoproject.org/cgit/cgit.cgi/meta-raspberrypi/ layer

**Describe the results you received:**
no IP Address

**Describe the results you expected:**
System with IP-Address

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
sumo
DISTRO 
poky 
DL_DIR 
/build/downloads 
IMAGE_FSTYPES 
rpi-sdimg tar.bz2 ext4 
IMAGE_INSTALL_append 
kernel-modules  
PACKAGE_CLASSES 
package_rpm 
SSTATE_DIR 
/build/sstate-cache 
PREFERRED_PROVIDER_virtual/kernel
linux-raspberrypi-dev 
RPI_USE_U_BOOT
1

GreSuh, 2018-05-13:
same problem with default kernel

pabigot, 2018-05-14:
I suspect a problem with the startup infrastructure.  When I boot the current sumo eth0 works fine, but the MAC address is not the one assigned to the device.  `/proc/cpuinfo` on other machines provides a serial number from which network addresses are derived; this is not present in the raspberrypi-64 build.  Interestingly the wlan0 mac address is correct.

Check whether the mac address you get on boot matches what you expect in your DHCP server.

pabigot, 2018-05-14:
This is probably #253.

GreSuh, 2018-05-14:
> This is probably #253.
In my setup i updated the machine definition to match raspberry3 device list.

agherzan, 2018-05-14:
I hit the same issue lately. It should get fixed by @pabigot 's PR.

agherzan, 2018-05-14:
@GreSuh this issue should be fixed now in sumo and master branches. Please give it a try.

closed 2018-05-14 by agherzan

GreSuh, 2018-05-14:
Thanks for todays changes on sumo. Network seems to work now. :)

agherzan, 2018-05-14:
@GreSuh we owe @pabigot a :beer: