agherzan/meta-raspberrypi #258: userland

ikoes, 2018-05-15:
hello guys,

I'm using userland from meta-raperrypi/recipe-graphics@master to build libopenmaxil.so.

Since poky_default_distro_feature sets "wayland", the userland recipe will set pkcgconfig[wayland]

After the image is generated I use gstreamer to encode a video source and I get the following error:
` ERROR                    omx gstomx.c:159:gst_omx_core_acquire: Failed to load module '/usr/lib/libopenmaxil.so': /usr/lib/libbrcmEGL.so: undefined symbol: wl_callback_interface`

If I try
```

# nm -D /usr/lib/libbrcmEGL.so
    U wl_callback_interface

# nm -D /usr/lib/libwayland-client.so
    00018e50 D wl_callback_interface

# nm -D /usr/lib/libwayland-server.so
    0001ae50 D wl_callback_interface
```

Does anybody know how to fix it ?

Thanks,

F.

kraj, 2018-05-19:
I thought we fixed this issue already. Problem is its not linking with wayland libraries, I will see if I can reproduce it

agherzan, 2018-07-26:
@kraj Did you manage to reproduce this?

kraj, 2018-07-26:
Not yet, keep it open.

kraj, 2018-11-28:
I think this should not be happening anymore, I dumped the DT_NEEDED section of libbrcmEGL.so and I see both libwayland-client.so.0 and libwayland-server.so.0 there so the needed symbol would be supplied

```
root@raspberrypi3:~# readelf -d /usr/lib/libbrcmEGL.so

Dynamic section at offset 0x1c958 contains 30 entries:
  Tag        Type                         Name/Value
 0x00000001 (NEEDED)                     Shared library: [libbcm_host.so]
 0x00000001 (NEEDED)                     Shared library: [libwayland-client.so.0]
 0x00000001 (NEEDED)                     Shared library: [libwayland-server.so.0]
 0x00000001 (NEEDED)                     Shared library: [libc.so]
 0x00000001 (NEEDED)                     Shared library: [libvchostif.so]
 0x00000001 (NEEDED)                     Shared library: [libvchiq_arm.so]
 0x00000001 (NEEDED)                     Shared library: [libvcos.so]
 0x0000000e (SONAME)                     Library soname: [libbrcmEGL.so]
```

@ikoes can you retest your case with latest master/thud

kraj, 2018-12-15:
I don't see this issue anymore, I am going to close it. Please re-open if this re-occurs with master and then we can revisit

closed 2018-12-15 by kraj