agherzan/meta-raspberrypi #260: linux-firmware licence issue

ikoes, 2018-05-16:
Hello guys,

I'm using meta-raspberrypi@rocko. Since I need the firmware or the raspberri3b+ I've taken the linux-firmware recipe from the actual master branch.

When I try to bitbake it I get the following error:
` linux-firmware-1_0.0+gitAUTOINC+bf04291309+86e88fbf03+e28cd7ee86-r0 do_populate_lic: Could not copy license file /home/worker/iagocto/build/tmp/work/all-poky-linux/linux-firmware/1_0.0+gitAUTOINC+bf04291309+86e88fbf03+e28cd7ee86-r0/git/LICENCE.cypress to /home/worker/iagocto/build/tmp/work/all-poky-linux/linux-firmware/1_0.0+gitAUTOINC+bf04291309+86e88fbf03+e28cd7ee86-r0/license-destdir/linux-firmware/generic_Firmware-cypress: [Errno 2] No such file or directory: '/home/worker/iagocto/build/tmp/work/all-poky-linux/linux-firmware/1_0.0+gitAUTOINC+bf04291309+86e88fbf03+e28cd7ee86-r0/git/LICENCE.cypress'`

pabigot, 2018-05-16:
You need to update your poky/oe-core to the current head of whichever release you're using (sumo or master).  The linux-firmware version you're using (`bf04291309` on 2017-10-09) pre-dates the addition of the Cypress license (`7f93c9deb4` on 2017-12-06).

pabigot, 2018-05-16:
The linux-firmware version you're using (bf04291309 on 2017-10-09) pre-dates the addition of the Cypress license (7f93c9deb4 on 2017-12-06).  Rocko's version pre-dates that.  You can try a bbappend to bring SRCREV up to what's in sumo or master, both of which work.

However, I think there was comment elsewhere that 3B+ won't work on rocko, so I suggest updating to sumo.

agherzan, 2018-05-16:
@ikoes we don't actively work on rocko anymore so the first recommendation would be for you to update to sumo. Otherwise, we count on you in contributing the backports to rocko as needed.

agherzan, 2018-07-26:
Closing due to no activity. We can re-open it if needed.

closed 2018-07-26 by agherzan