agherzan/meta-raspberrypi #262: netopeer encountering alignment error when starting service

cclark1974, 2018-05-22:
I pulled the latest changes and Im still seeing the alignment errors when starting netopeer/sysrepo service.

May 19 03:26:54 raspberrypi3 daemon.info init: Id "S0" respawning too fast: disabled for 5 minutes                                                        

May 19 03:28:10 raspberrypi3 user.warn kernel: [26307.707117] Alignment trap: sysrepod (13191) PC=0x76e54c5c Instr=0xe8ae000f Address=0x7490563d FSR 0x801             

May 19 03:28:10 raspberrypi3 user.warn kernel: [26307.708698] Alignment trap: sysrepod (13191) PC=0x76e54c64 Instr=0xe8ae000f Address=0x7490564d FSR 0x801

May 19 03:28:10 raspberrypi3 user.warn kernel: [26307.710290] Alignment trap: sysrepod (13191) PC=0x76e54c6c Instr=0xe88e000f Address=0x7490565d FSR 0x801

May 19 03:28:10 raspberrypi3 user.warn kernel: [26307.712484] Alignment trap: sysrepod (13191) PC=0x76e53e84 Instr=0xe1c020f0 Address=0x74905711 FSR 0x801

May 19 03:28:10 raspberrypi3 user.warn kernel: [26307.714229] Alignment trap: sysrepod (13191) PC=0x76e71bdc Instr=0xedc70b0a Address=0x74905665 FSR 0x801

May 19 03:28:10 raspberrypi3 user.err kernel: [26307.716033] Alignment trap: not handling instruction edc70b0a at [<76e71bdc>]              

May 19 03:28:10 raspberrypi3 user.alert kernel: [26307.718002] Unhandled fault: alignment exception (0x801) at 0x74905665                   

May 19 03:28:10 raspberrypi3 user.alert kernel: [26307.720148] pgd = b7adc000                                                               

May 19 03:28:10 raspberrypi3 user.alert kernel: [26307.722442] [74905665] *pgd=37802835, *pte=36f6f75f, *ppte=36f6fc7f                      

May 19 03:28:10 raspberrypi3 daemon.err netopeer2-server[13197]: Unable to get list of schemas supported by sysrepo (The peer disconnected).

May 19 03:28:10 raspberrypi3 daemon.err netopeer2-server[13197]: Server init failed.                             

May 19 03:28:10 raspberrypi3 daemon.err netopeer2-server[13197]: nc_ps_clear: invalid argument (ps).

Im running master branch on the meta-raspberrypi 

I built on raspberry pi3, 32b arch.

Any further info needed please let me know

betafive, 2018-05-22:
Hi, you say you're "still seeing" this error. Is there any previous discussion of this? It's not something I've seen before but I might have just missed it.

agherzan, 2018-05-22:
@cclark1974 please update the description in such a way that it can provide context, build description, sources versions and ways to reproduce. The issue template has some content for a reason.

cclark1974, 2018-05-23:
Local.conf:
#
# This file is your local configuration file and is where all local user settings
# are placed. The comments in this file give some guide to the options a new user
# to the system might want to change but pretty much any configuration option can
# be set in this file. More adventurous users can look at local.conf.extended
# which contains other examples of configuration which can be placed in this file
# but new users likely won't need any of them initially.
#
# Lines starting with the '#' character are commented out and in some cases the
# default values are provided as comments to show people example syntax. Enabling
# the option is a question of removing the # character and making any change to the
# variable as required.

#
# Machine Selection
#
# You need to select a specific machine to target the build with. There are a selection
# of emulated machines available which can boot and run in the QEMU emulator:
#
#MACHINE ?= "qemuarm"
#MACHINE ?= "qemuarm64"
#MACHINE ?= "qemumips"
#MACHINE ?= "qemumips64"
#MACHINE ?= "qemuppc"
#MACHINE ?= "qemux86"
#MACHINE ?= "qemux86-64"
#
# There are also the following hardware board target machines included for 
# demonstration purposes:
#
#MACHINE ?= "beaglebone-yocto"
#MACHINE ?= "genericx86"
#MACHINE ?= "genericx86-64"
#MACHINE ?= "mpc8315e-rdb"
#MACHINE ?= "edgerouter"
#
# This sets the default machine to be qemux86 if no other machine is selected:
#MACHINE ??= "qemux86"
MACHINE ??= "raspberrypi3"

#
# Where to place downloads
#
# During a first build the system will download many different source code tarballs
# from various upstream projects. This can take a while, particularly if your network
# connection is slow. These are all stored in DL_DIR. When wiping and rebuilding you
# can preserve this directory to speed up this part of subsequent builds. This directory
# is safe to share between multiple builds on the same machine too.
#
# The default is a downloads directory under TOPDIR which is the build directory.
#
#DL_DIR ?= "${TOPDIR}/downloads"
#IMAGE_FSTYPES += "wic.qcow2"

#
# Where to place shared-state files
#
# BitBake has the capability to accelerate builds based on previously built output.
# This is done using "shared state" files which can be thought of as cache objects
# and this option determines where those files are placed.
#
# You can wipe out TMPDIR leaving this directory intact and the build would regenerate
# from these files if no changes were made to the configuration. If changes were made
# to the configuration, only shared state files where the state was still valid would
# be used (done using checksums).
#
# The default is a sstate-cache directory under TOPDIR.
#
#SSTATE_DIR ?= "${TOPDIR}/sstate-cache"
[... 227 lines cut ...]
Build Configuration:
BB_VERSION           = "1.37.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal-4.8"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.5"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:5479654eeaaa0f81bfff54ca49369c87f1658705"
meta-oe              
meta-networking      
meta-webserver       
meta-python          
meta-multimedia      = "master:a19aa29f7fa336cd075b72c496fe1102e6e5422b"
meta-raspberrypi     = "master:1fb4dcc0b57bb51aac9c047f17048e1121b690fb"
meta-sysrepo         = "master:f966edff8a88cf0834b55cb151d2cb3443ba53a6"

after booting the raspberry pi with the img, I attempt to start netopeer by:
/etc/init.d/netopeer start 

that is when I encounter this error.

cclark1974, 2018-05-23:
bblayers:
NOTE: Starting bitbake server...
layer                 path                                      priority
==========================================================================
meta                  /root/Yocto/poky/meta                     5
meta-poky             /root/Yocto/poky/meta-poky                5
meta-yocto-bsp        /root/Yocto/poky/meta-yocto-bsp           5
meta-oe               /root/Yocto/poky/meta-openembedded/meta-oe  6
meta-networking       /root/Yocto/poky/meta-openembedded/meta-networking  5
meta-webserver        /root/Yocto/poky/meta-openembedded/meta-webserver  6
meta-python           /root/Yocto/poky/meta-openembedded/meta-python  7
meta-multimedia       /root/Yocto/poky/meta-openembedded/meta-multimedia  6
meta-raspberrypi      /root/Yocto/poky/meta-raspberrypi         9
meta-sysrepo          /root/Yocto/poky/meta-sysrepo             6

hhromic, 2018-05-23:
hi @cclark1974 , can you please use [fenced code-blocks](https://help.github.com/articles/creating-and-highlighting-code-blocks/#fenced-code-blocks) for your pasted information/files so they are *much more* readable here in github? Thanks :)

cclark1974, 2018-05-24:
any updates on this issue?  is there more info needed?

betafive, 2018-06-01:
I just had a quick look around at the `meta-sysrepo` layer and at `sysrepo` itself. Perhaps this is related to https://github.com/sysrepo/sysrepo/issues/947 ? You may need to modify the config options in the sysrepo recipe to disable the feature that's discussed in that issue.

agherzan, 2018-06-01:
The above-mentioned issue is the only guess I have as well. I would recommend you to either add additional information to that bug or report it to rpi kernel repository. As far as I can tell at this point it doesn't sound like an issue with the yocto layer.

pbrkr, 2019-11-14:
@agherzan recommend we close this issue now as there's been no update in over a year.

closed 2019-11-15 by agherzan