agherzan/meta-raspberrypi #265: Build poky 2.5 using meta-raspberrypi. Issues with rpi-config_git.bb

ftchopfambie, 2018-06-20:
**Description**

**Steps to reproduce the issue:**
1. Download poky 2.5 git clone -b sumo https://git.yoctoproject.org/git/poky
2. Download meta-raspberrypi
3. source the poky and add  IMAGE_INSTALL_append = "rpi-config" in the build/conf/local.conf with all the other configurations
4. build the image: bitbake core-image-full-cmdline

**Describe the results you received:**
ERROR: Task do_populate_sdk in /home/fabrice/SeparatorRS-yocto/poky/../meta-SeparatorRS/recipes-core/images/separatorrs-image.bb rdepends upon non-existent task do_package_write_ipk in /home/fabrice/SeparatorRS-yocto/poky/../meta-raspberrypi/recipes-bsp/bootfiles/rpi-config_git.bb

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
revisions meta-raspberrypi sumo, poky sumo
distro poky 2.5

agherzan, 2018-06-20:
The error you are pasting above is not while running the build for `core-image-full-cmdline` as you describe in the steps to reproduce. It for `separatorrs-image` which I assume is a custom one. Anyway, I did try to reproduce with `core-image-full-cmdline` and could see the problem. I'll take a look to see why this happens.

agherzan, 2018-06-20:
Right. The issue comes from the fact that you are trying to "install" a recipe that doesn't generate any packages. `rpi-config` is a recipe which only populates something in the deploy directory. It doesn't create any rpms, ipks etc. That's why only the image generation CMD depends on it (see do_image_rpi_sdimg[depends]) and this is done because that image generation handles the boot partitions as well.

Hope this clarifies this.

closed 2018-06-20 by agherzan

ftchopfambie, 2018-06-21:
So this means, I don't really need to add rpi-config as packet to install while building an image for raspberrypi, right? Cause really was divided about the fact that I needed it. I remove the inherit no packages from rpi-config_git.bb and it did work.

agherzan, 2018-06-27:
Thatis true. You don't need that package in the rootfs.