agherzan/meta-raspberrypi #273: userland: new patch cause qtbase build fail

myke-80, 2018-06-25:
The commit [userland: bumped version and backport khronos](https://github.com/agherzan/meta-raspberrypi/commit/237e0c79c694d85d3eaf37cebe386939f6c0257d?diff=split) introduced a [patch](https://github.com/agherzan/meta-raspberrypi/commit/237e0c79c694d85d3eaf37cebe386939f6c0257d?diff=split#diff-dc80539332edfdd76c2297f1bd34d346) that cause the qtbase build failure.

**Steps to reproduce the issue:**
1. Added qtbase_git.bbappend with packages "accessibility eglfs fontconfig gles2 linuxfb tslib" and depend on userland
2. added a generic qt project to be built in the image ([this](https://github.com/myke-80/qt-quick-test) for instance)
3. qtbase fails, compile log attached ('EGLStreamKHR' has not been declared)

[log.do_compile.zip](https://github.com/agherzan/meta-raspberrypi/files/2132622/log.do_compile.zip)

betafive, 2018-06-25:
We need some more information to be able to help here. Could you let us know which layers and branches you're using, what modifications you've made (including modifications in local.conf), the full bitbake command you ran and the full output.

hhromic, 2018-06-25:
My mistake. In retrospective, I think the define guard I used (`EGL_KHR_stream`) is causing this problem.
I backported that from upstream Khronos for the typdef, but it is not actually backporting the full EGL_KHR_stream feature from Khronos.
By using that define guard as-is, gtbase seems to be trying to compile against this non-existing feature in the outdated userland Khronos. I think the right thing to do is instead to use a non-clashing define guard name specific for this patch (bringing the typedef).
What do you think @betafive?

myke-80, 2018-06-25:
Sure, I will post the result as soon as finished.
Thanks !

myke-80, 2018-06-25:
@hhromic : compiled and succeeded. I'm not able to try it on a device in this moment, but I think is going to be fine.

hhromic, 2018-06-25:
Thanks for the feedback. I will review this patch then to account for this issue and the original issue being patched. Sorry for the inconvenience!

myke-80, 2018-06-25:
No problem, thanks for looking immediately !

agherzan, 2018-06-27:
Looks good to me @hhromic . Merged to master.

PR #274 (2018-06-27): userland: fix backport khronos typedef patch
The patch in 237e0c7 used a define guard that clashes with an actual feature of Khronos (EGL_KHR_stream).
This makes some packages such as `qtbase` to think that the outdated Krhonos in userland has newer features.
This fix uses a more specific name for the relevant define guard (EGL_KHR_uint64_typedef).

Fixes #273
closed 2018-06-27 by agherzan