agherzan/meta-raspberrypi #275: Question about image types

michaelbaisch, 2018-07-01:
Hello,

by default the `rpi-sdimg` image is built which is defined a separate class. But one could also use the`wic` option. Since I wanted to add another partition to my build I tried the `wic` option and it also seems to work. I can still `dd` on the SD card, like I could before. So why are there two options? Why isn't there just the `wic` option, where one could easily add for example partitions or similar things? Should I be aware of things that are now different in my build?

Thanks,
Michael

agherzan, 2018-07-07:
When the sdimg image type was implemented there was no wic in place. Generally, sdimg gave us more flexibility and I, personally, still use it as the main image generation tool. So at this point, I'd say use whatever makes you feel more comfortable.

michaelbaisch, 2018-07-07:
Alright thanks :)
Good, I don't have to have a bad feeling using wic. Is there any way to get `bootloader --append=` to work?

agherzan, 2018-07-07:
That sounds like a question for oe-core. If you figure out something wrong in our configuration, ping me.

closed 2018-07-07 by agherzan

michaelbaisch, 2018-07-08:
I assume it's something wrong with the raspberry layer. For another non-raspberry build I can use the `bootloader --append=` without trouble. I suspect it's because of the custom bootloader, maybe there is nothing we could to about it. Just want to through it out there, because it would make separating filesystem and kernel stuff easier. Right now I'm using the `CMDLINE_append` variable in a kernel append recipe for filesystem stuff.

The [wic help](https://github.com/alexandrebelloni/wic/blob/master/scripts/lib/image/help.py) it states:

```
* bootloader

      This command allows the user to specify various bootloader
      options.  The following are supported 'bootloader' options:
      
        --timeout: Specifies the number of seconds before the
                   bootloader times out and boots the default option.

        --append: Specifies kernel parameters. These will be added to
                  bootloader command-line - for example, the syslinux
                  APPEND or grub kernel command line.

      Note that bootloader functionality and boot partitions are
      implemented by the various --source plugins that implement
      bootloader functionality; the bootloader command essentially
      provides a means of modifying bootloader configuration.
```

kraj, 2018-07-08:
you might have to write a wic bootloader source plugin for rpi bootloader.