agherzan/meta-raspberrypi #282: Q: CM3 inherited from RPI2 and not RPI3?

nandra, 2018-07-11:
I'm trying to use CM3 with eMMC with meta-raspberrypi and it boots fine with build image. Interestingly when looking at documentation for CM3 it should be similar to RPI3 (processor used is: https://www.raspberrypi.org/documentation/hardware/raspberrypi/bcm2837/README.md). Even if it is working shouldn't be cm3 inherited from RPI3-64? Sorry for stupid question ;). Marek

kraj, 2018-07-11:
we have separate machine configs for 64bit RPI 3 machines, but I think you are right that it should be inheriting raspberrypi3 machine setup and not rpi2 as far as SOC is concerned @agherzan added this machine config for cm3 and I guess there might have been some reason to pin it to rpi2 and not use rpi3 specs.

nandra, 2018-07-11:
IIRC CM3 should use same cpu ase RPI 3 - 64 bit. So probably inherit from raspberry3-64? My problem is I'm hunting is that CM# have eMMC but in u-boot it's not working so I'm trying to find root cause. Thanks for heads up.

kraj, 2018-07-11:
first try the 32bit version.

nandra, 2018-07-12:
I tried with 32bit version and eMMC start working. But I didn't try u-boot-2017.09 from rocky but latest 2018.07 which works fine. I'll try also with 2017.09 and let you know.

kraj, 2018-07-13:
OK thanks,  would you care to send a pull request to change the SOC type in machine conf file.

nandra, 2018-07-13:
Yes sure.

nandra, 2018-07-16:
@kraj well situation is bit problematic. Inherit from raspberry3 is OK but with this setup in rocko u-boot won't start at all (I need to use latest one). So I'll create PR only for machine change then.

agherzan, 2018-07-16:
@nandra @kraj I can't remember exactly why this happened but I think it was from the times where rpi3 and cm3 where inheriting rpi2. We should change it to rpi3. @nandra What do you mean by the latest one?

agherzan, 2018-07-16:
@nandra https://github.com/agherzan/meta-raspberrypi/pull/288

agherzan, 2018-07-16:
Fixed by #288.

closed 2018-07-16 by agherzan