agherzan/meta-raspberrypi #283: rpi-hwup-image-raspberrypi3.rpi-sdimg login connection issue

becklo, 2018-07-11:
Hi,

I have manage to build a Raspberry Pi 3 image using :
_bitbake rpi-hwup-image_  
The image boot from the image created with the previous command. Then it ask for a login and password.
I tried root as login without password but it doesn't seems to be the right combination... 

Is there something obvious that I am missing ?

Best regards,
becklo

kraj, 2018-07-11:
@becklo  this image is deprecated, core-image-minimal should be used now instead if you are not able to login that means you may not be having ```debug-tweaks``` in ```IMAGE_FEATURES```

you could do so by adding below line to local.conf

```
EXTRA_IMAGE_FEATURES = "debug-tweaks"
```

becklo, 2018-07-12:
I have added this line in my local.conf and created a new image with core-image-minimal but now it doesn't boot... 
I am using a screen connected via HDMI to check if it boot. How is it possible ? 
Here is my local.conf :

> DISTRO_FEATURES_append = " systemd"
> VIRTUAL-RUNTIME_init_manager = "systemd"
> DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
> VIRTUAL-RUNTIME_initscripts = ""
> 
> ENABLE_UART = "1"
> 
> KERNEL_IMAGETYPE = "uImage"
> RPI_USE_U_BOOT = "1"
> MACHINE ?= "raspberrypi3"
> EXTRA_IMAGE_FEATURES = "debug-tweaks"

agherzan, 2018-07-16:
@becklo Can you attach a serial console? Nothing attaches to your HDMI in that image you are using.

agherzan, 2018-07-20:
Closing for now. It looks like a wrong configuration.

closed 2018-07-20 by agherzan