agherzan/meta-raspberrypi #284: bcm2710-rpi-cm3.dtb missing from KERNEL_DEVICETREE for raspberrypi3-64 in sumo

stevep-korg, 2018-07-11:
The file bcm2710-rpi-cm3.dtb is missing from the KERNEL_DEVICETREE list for the raspberrypi3-64 machine target in sumo branch.

Unable to boot a Raspberry Pi Compute Module 3 using 64-bit poky Linux targeted for raspberrypi3-64 built in sumo branch.

**Steps to reproduce the issue:**
1.  Build poky Linux for raspberrypi3-64 machine target, at head of sumo
2. Copy to eMMC on Compute Module 3 and try to boot it

It gets stuck on the rainbow screen, because the bcm2710-rpi-cm3.dtb file was not placed in the boot partition of the image during the build.

The Compute Module 3 should boot fine, continuing past the rainbow screen, and on to the rest of the boot.

The solution is to list bcm2710-rpi-cm3.dtb in the KERNEL_DEVICETREE listing in raspberrypi3-64.conf, in meta-raspberrypi/conf/machine

This problem was observed in the sumo branch.

agherzan, 2018-07-16:
@stevep-korg This should do it: https://github.com/agherzan/meta-raspberrypi/pull/289

agherzan, 2018-07-16:
Fixed by #289 .

closed 2018-07-16 by agherzan

stevep-korg, 2018-07-16:
Thanks very much, Andrei! What is the process for such fixes being merged into the other branches like sumo?  (I work in sumo branch, so I won't currently see this fix there, I don't think.)

stevep-korg, 2018-07-16:
Andrei, we may need to enhance this fix soon, or temporarily back it off.  I just discovered this error when it was trying to use that change locally:

|   CHK     scripts/mod/devicetable-offsets.h
| make[3]: *** No rule to make target 'arch/arm64/boot/dts/broadcom/bcm2710-rpi-cm3.dtb'.  Stop.
| arch/arm64/Makefile:133: recipe for target 'broadcom/bcm2710-rpi-cm3.dtb' failed

Perhaps there is also another issue with a missing dts or dts entry and therefore this in not resolving.

Sorry for this complication.  I thought I had tested this and seen that it would build OK before suggesting the change.

Please let me know your thoughts.

stevep-korg, 2018-07-16:
This additional problem must stem from the lack of the bcm2710-rpi-cm3.dts in the arch/arm64/boot/dts/broadcom/dts folder.  That's unfortunate.

stevep-korg, 2018-07-18:
> The usual process is that people who need them would PR the fixes on top of stable branches. So I encourage you to push this fix on top of sumo and PR it here.

Hi Andrei,

I don’t know how to do this process, because I’m new to it.

Should I clone the repo in my github account and do some work there?

Sorry, I don’t understand the workflow of how to achieve this.  Is there a link with some directions about how to do this?

(I guess PR is Pull Request.)

Another problem is that the companion dts file is also missing from sumo, so another fix is needed in sumo besides putting the dtb in the KERNEL_DEVICETREE list.

Thanks,
Steve