agherzan/meta-raspberrypi #285: IMAGE_BOOT_FILES broken after recent kernel-devicetree.bbclass changes

jonathanhaigh-arm, 2018-07-12:
**Description**
The recent openembedded-core commit [1860d9d3c62e2e94cd68a809385873ffd8270b6d](https://github.com/openembedded/openembedded-core/commit/1860d9d3c62e2e94cd68a809385873ffd8270b6d) changed the names of DTB files in the deploy directory, but meta-raspberrypi's IMAGE_BOOT_FILES content hasn't been updated to reflect this. 

The DTB related contents of IMAGE_BOOT_FILES is generated using the make_dtb_boot_files function in conf/machine/include/rpi-base.inc which expects the file names of the DTBs in the deploy directory to be prefixed with ${KERNEL_IMAGETYPE}, but they no longer are.

I'd guess that this won't affect users of the sdcard_image-rpi class due to https://github.com/agherzan/meta-raspberrypi/commit/4826d225807850f082d99ba42019023e54215b84, but it does break my build that uses Wic.

**Steps to reproduce the issue:**
1. Build an image with MACHINE=raspberrypi3 and "wic.gz" in IMAGE_FSTYPES. 

**Describe the results you received:**

```
02:51:55 ERROR: my-image-1.0-r0 do_image_wic: Function failed: do_image_wic (log file is located at /work/rpi3/build/tmp/work/raspberrypi3-oe-linux-gnueabi/my-image/1.0-r0/temp/log.do_image_wic.31224)
02:51:55 ERROR: Logfile of failure stored in: /work/rpi3/build/tmp/work/raspberrypi3-oe-linux-gnueabi/my-image/1.0-r0/temp/log.do_image_wic.31224
02:51:55 Log data follows:
```
...
```
02:51:55 | DEBUG: Executing shell function do_image_wic
02:51:55 | INFO: Creating image(s)...
02:51:55 | 
02:51:55 | ERROR: _exec_cmd: install -m 0644 -D /work/rpi3/build/tmp/deploy/images/raspberrypi3/uImage-bcm2708-rpi-0-w.dtb /work/rpi3/build/tmp/work/raspberrypi3-oe-linux-gnueabi/my-image/1.0-r0/deploy-my-image-image-complete/my-image-raspberrypi3-20180711002549/tmp.wic.qc8r2h_s/boot.1/bcm2708-rpi-0-w.dtb returned '1' instead of 0
02:51:55 | output: install: cannot stat '/work/rpi3/build/tmp/deploy/images/raspberrypi3/uImage-bcm2708-rpi-0-w.dtb': No such file or directory
02:51:55 | 
02:51:55 | WARNING: exit code 1 from a shell command.
02:51:55 | ERROR: Function failed: do_image_wic (log file is located at /work/rpi3/build/tmp/work/raspberrypi3-oe-linux-gnueabi/my-image/1.0-r0/temp/log.do_image_wic.31224)
02:51:55 NOTE: recipe my-image-1.0-r0: task do_image_wic: Failed
```

**Describe the results you expected:**
No build errors.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**
meta-rpi version: efa848ca84a5ed541d4a755d19e66855214bfa23
oe-core version: f9324af88a99eca28b160fa31aa4516fd397e44b

hhromic, 2018-07-12:
I can confirm as a user of the `sdcard_image-rpi` class that this doesn't affect it.

agherzan, 2018-07-16:
@jonathanhaigh-arm Is the current master HEAD still reproducing this issue?

jonathanhaigh-arm, 2018-07-17:
I'm building again now...

jonathanhaigh-arm, 2018-07-17:
... and it still fails in the same way. I'll try to come up with a PR.

jonathanhaigh-arm, 2018-07-17:
Created PR https://github.com/agherzan/meta-raspberrypi/pull/290

agherzan, 2018-07-20:
PR merged. Closing this.

closed 2018-07-20 by agherzan