agherzan/meta-raspberrypi #286: vc4graphics + wayland + HDMI  on rocko produces OOM errors among other things

mirzak, 2018-07-13:
**Description**

I am currently doing some work that involves the components that are mentioned in the title. But to repeat we are trying to run an Qt application using wayland/weston backend on a raspberrypi3 board. 

We are using `vc4graphics` instead of `userland` and a display connected on HDMI with touch support over USB. The graphics work as expected initially.

But there are multiple issues here, first one is this:

```
[  414.810574] WARNING: CPU: 1 PID: 394 at /kernel-source//kernel/irq/manage.c:527 __enable_irq+0x54/0x80
[  414.820018] Unbalanced enable for IRQ 40
[  414.823993] Modules linked in: rfcomm bnep hci_uart btbcm ecryptfs bluetooth panel_raspberrypi_touchscreen brcmfmac brcmutil cfg80211 rfkill vc4 snd_soc_core snd_compress i2c_gpio snd_pcm_dmaengine bcm2835_gpi
omem uio_pdrv_genirq uio fixed evdev joydev sch_fq_codel snd_bcm2835 snd_pcm snd_timer snd hid_multitouch fuse ipv6
[  414.853179] CPU: 1 PID: 394 Comm: QSGRenderThread Tainted: G        W       4.9.80 #1
[  414.861123] Hardware name: BCM2835
[  414.864592] [<8011102c>] (unwind_backtrace) from [<8010c8f0>] (show_stack+0x20/0x24)
[  414.872458] [<8010c8f0>] (show_stack) from [<8045af48>] (dump_stack+0xcc/0x110)
[  414.879886] [<8045af48>] (dump_stack) from [<8011e7fc>] (__warn+0xf8/0x110)
[  414.886956] [<8011e7fc>] (__warn) from [<8011e85c>] (warn_slowpath_fmt+0x48/0x50)
[  414.894556] [<8011e85c>] (warn_slowpath_fmt) from [<8017752c>] (__enable_irq+0x54/0x80)
[  414.902685] [<8017752c>] (__enable_irq) from [<8017759c>] (enable_irq+0x44/0x7c)
[  414.910283] [<8017759c>] (enable_irq) from [<7f1c2b64>] (vc4_irq_postinstall+0x24/0x40 [vc4])
[  414.919062] [<7f1c2b64>] (vc4_irq_postinstall [vc4]) from [<7f1c5858>] (vc4_v3d_runtime_resume+0x58/0x60 [vc4])
[  414.929368] [<7f1c5858>] (vc4_v3d_runtime_resume [vc4]) from [<8053eb74>] (pm_generic_runtime_resume+0x3c/0x48)
[  414.939617] [<8053eb74>] (pm_generic_runtime_resume) from [<80542a28>] (__genpd_runtime_resume+0x3c/0xc0)
[  414.949334] [<80542a28>] (__genpd_runtime_resume) from [<80544638>] (genpd_runtime_resume+0x84/0x198)
[  414.958699] [<80544638>] (genpd_runtime_resume) from [<80540140>] (__rpm_callback+0x48/0x98)
[  414.967271] [<80540140>] (__rpm_callback) from [<805401c0>] (rpm_callback+0x30/0x90)
[  414.975138] [<805401c0>] (rpm_callback) from [<805415ec>] (rpm_resume+0x500/0x7c0)
[  414.982827] [<805415ec>] (rpm_resume) from [<8054191c>] (__pm_runtime_resume+0x70/0x9c)
[  414.991025] [<8054191c>] (__pm_runtime_resume) from [<7f1bf6c4>] (vc4_submit_cl_ioctl+0x2e4/0x9cc [vc4])
[  415.000716] [<7f1bf6c4>] (vc4_submit_cl_ioctl [vc4]) from [<80511dfc>] (drm_ioctl+0x1f0/0x3f8)
[  415.009466] [<80511dfc>] (drm_ioctl) from [<802867a4>] (do_vfs_ioctl+0xb0/0x814)
[  415.016981] [<802867a4>] (do_vfs_ioctl) from [<80286f4c>] (SyS_ioctl+0x44/0x68)
[  415.024410] [<80286f4c>] (SyS_ioctl) from [<801082a0>] (ret_fast_syscall+0x0/0x1c)
[  415.032092] ---[ end trace 3f3b9a47efa1f1f8 ]---
```
Above is printed on any interaction on the display, e.g when I click an icon or change view.

The second problem and more serious is this:

```
[  416.093734] vc4-drm soc:gpu: failed to allocate buffer with size 8355840
[  416.100782] vc4-drm soc:gpu: failed to allocate buffer with size 8355840
[  416.107807] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  416.114878] [drm]                         kernel:   8100kb BOs (1)
[  416.121159] [drm]                            V3D: 222088kb BOs (121)
[  416.127655] [drm]                     V3D shader:    212kb BOs (53)
[  416.134041] [drm]                           dumb:     32kb BOs (2)
[  416.140316] [drm]                         binner:  16384kb BOs (1)
[  416.146979] vc4-drm soc:gpu: failed to allocate buffer with size 8355840
[  416.287439] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.294543] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.301541] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  416.308578] [drm]                         kernel:   8100kb BOs (1)
[  416.314889] [drm]                            V3D: 230076kb BOs (118)
[  416.321344] [drm]                     V3D shader:    236kb BOs (59)
[  416.327735] [drm]                           dumb:     32kb BOs (2)
[  416.334275] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.343140] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.350195] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  416.357244] [drm]                         kernel:   8100kb BOs (1)
[  416.363523] [drm]                            V3D: 221916kb BOs (117)
[  416.370014] [drm]                     V3D shader:    236kb BOs (59)
[  416.376419] [drm]                           dumb:     32kb BOs (2)
[  416.383090] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.390239] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.397292] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  416.404322] [drm]                         kernel:   8100kb BOs (1)
[  416.410599] [drm]                            V3D: 221916kb BOs (117)
[  416.417082] [drm]                     V3D shader:    236kb BOs (59)
[  416.423446] [drm]                           dumb:     32kb BOs (2)
[  416.429761] vc4_v3d 3fc00000.v3d: Failed to allocate memory for tile binning: -12. You may need to enable CMA or give it more memory.
[  416.443280] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.450462] vc4-drm soc:gpu: failed to allocate buffer with size 16777216
[  416.457526] [drm:vc4_bo_create [vc4]] *ERROR* Failed to allocate from CMA:
[  416.464556] [drm]                         kernel:   8100kb BOs (1)
[  416.470832] [drm]                            V3D: 221916kb BOs (117)
[  416.477319] [drm]                     V3D shader:    236kb BOs (59)
[  416.483729] [drm]                           dumb:     32kb BOs (2)
```
Which crashes the application that is running.

I tried updating the Linux kernel on the rocko branch to 4.14, this actually got rid of problem number one, but the OOM problem persisted.

I am running the same configuration and application on other hardware platform which indicates that it is not an application issue.

My configuration on the raspberrypi:

cmdline.txt
```
dwc_otg.lpm_enable=0 console=serial0,115200 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
```

config.txt
```
# Enable UART
enable_uart=1
# Enable VC4 Graphics
dtoverlay=vc4-kms-v3d,cma-256
avoid_warnings=2
mask_gpu_interrupt0=0x400
dtoverlay=rpi-ft5406-overlay
dtparam=audio=on
```

/proc/cmdline
```
8250.nr_uarts=1 cma=256M bcm2708_fb.fbwidth=1920 bcm2708_fb.fbheight=1080 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyS0,115200
 root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
```
I am aware that I should probably report this issue on multiple locations but wanted to start out here to see if anyone else has successfully used the same combination that I am using.

Should also be noted that everything worked just fine on `pyro`. But there seems to have been some changes between `pyro` and `rocko` regarding how wayland/weston works on raspberrypi boards, e.g https://github.com/agherzan/meta-raspberrypi/commit/a1cfeb3324c458a73507e34ffcf28c25591b8052

Any insight are appreciated.

Thanks.

mirzak, 2018-07-13:
Similar issues reported here: https://archlinuxarm.org/forum/viewtopic.php?f=65&t=12665 (unresovled)

mirzak, 2018-07-13:
Update, tested using `sumo` of meta-raspberrypi while keeping the rest on `rocko`. Problems disappeared. That should narrow it down more.

mirzak, 2018-07-13:
@gunnarx <-- notification.

mirzak, 2018-07-13:
Also submitted this, https://github.com/raspberrypi/linux/pull/2618

hhromic, 2018-07-22:
this should be fixed by now?

mirzak, 2018-07-22:
Yepp, thanks for reminding.

Closing...

PR #287 (2018-07-20): Rocko fix wayland together with vc4
First commit fixes the OOM errors.

Second commit fixes Linux kernel warning "unbalanced IRQ enable" when using vc4

Fixes #286
closed 2018-07-22 by mirzak