agherzan/meta-raspberrypi #300: KERNEL_DEVICETREE cannot be extended

nandra, 2018-08-03:
I'm trying to add more dtbo to final image by specifying in linux-raspberrypi.bbappend or in local.conf by extending KEREL_DEVICETREE += "overlays/pwm.dtbo" but looks like when specify it in local.conf variable KERNEL_DEVICETREE is overwritten or when specify it in linux-raspberrypi.bbeappend it have no effect. Any ideas how to extend this variable to get more dtbo's on target? I workarounded it by extending variable in rpi-base.conf (but I would like to avoid that ;) ) Thanks.

agherzan, 2018-08-03:
Hi @nandra . Have you tried appending? We are doing something similar here: https://github.com/resin-os/resin-raspberrypi/blob/master/layers/meta-resin-raspberrypi/conf/layer.conf#L17. Appending using `_append`.

nandra, 2018-08-03:
I tried += also _append but it doesn't work. I'll re-check and report status (maybe my env gets broken somehow). Thanks.

nandra, 2018-08-29:
OK with _append it work which is quite strange because I assume that += also should work. Anyway when looking in rpi-base.inc we have 
```
KERNEL_DEVICETREE ?= " \

```
so softval is used while in 
raspberrypi3-64.conf there is:
```
KERNEL_DEVICETREE = " \
```

So maybe this could be a problem that when user want to add something to KERNEL_DEVICETREEE for non 64bit rpi then it cannot be overwritten? Thanks.

aswild, 2018-09-20:
- `local.conf` is parsed before the `${MACHINE}.conf`, so any sets with = in a machine (or distro) config will override `local.conf`
- Using += will clobber a default set with ?= (I'm not sure why bitbake behaves this way, but it does)
- Using _append should work around both of these (because _append is treated like an override and processed at the end of variable expansion)
- You need to modify KERNEL_DEVICETREE in a .conf file, not by bbappending the kernel recipe, because KERNEL_DEVICETREE is used in multiple recipes (the kernel but also the image recipes via sdcard_image-rpi.bbclass)

nandra, 2018-10-02:
@aswild thanks for detailed description. I think we can close this issue as _append works as expected. Thanks.

closed 2018-10-02 by nandra