agherzan/meta-raspberrypi #308: Zero W WiFi not working

MeantCoder, 2018-08-30:
Hello,

I'm compiled Yocto with linux-firmware-bcm43430 and distro feature wifi enabled but i can't use wifi on zero w.

iw list output : nl80211 not found.

nandra, 2018-10-02:
Looks like /lib/firmware is empty in final image (using sumo branch)

nandra, 2018-10-03:
I didn't have time to look on issue (why MACHINE_RRECOMENDS doesn't put firmware to rootfs). Anyway workaround is to add:
```
linux-firmware-bcm43430 \
linux-firmware-bcm43430a1-hcd \
```

to e.g. IMAGE_INSTALL (works for raspberrypi0-wifi only)

mirzak, 2018-10-03:
Did you mean MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS?

Which image did you build, because that will determinate if `MACHINE_ESSENTIAL_EXTRA_RRECOMMENDS` is included or not.

Basically the image must inherit from "core-image"for the above to be included.

nandra, 2018-10-03:
Nope I mean this one: https://github.com/agherzan/meta-raspberrypi/blob/2d40b000021bc8a9ef7f329ed0ad410f8d227b97/conf/machine/raspberrypi0-wifi.conf#L9

I build custom image (also core-image-minimal) and my custom image inherit core-image.

mirzak, 2018-10-03:
`core-image-minimal` will definitely not install `MACHINE_EXTRA_RRECOMMENDS`, even though it inherits `core-image.bbclass` it overrides the `IMAGE_INSTALL`

It looks like `MACHINE_EXTRA_RRECOMMENDS` is installed trough `packagegroup-base` which is is install by `core-image.bbclass` by including the `packagegroup-base-extended`.

Does your custom image "override" `IMAGE_INSTALL` or does it append to it. Because the logic in `core-image.bbclass` is:

```
CORE_IMAGE_BASE_INSTALL = '\
    packagegroup-core-boot \
    packagegroup-base-extended \
    \
    ${CORE_IMAGE_EXTRA_INSTALL} \
    '

CORE_IMAGE_EXTRA_INSTALL ?= ""

IMAGE_INSTALL ?= "${CORE_IMAGE_BASE_INSTALL}"
```

This means that if you have a `IMAGE_INSTALL = "something"` in your custom image then the base package-groups will not be installed.

Try building `core-image-base`.

nandra, 2018-10-03:
You're right @mirzak . I had in custom image IMAGE_INSTALL = instead of IMAGE_INSTALL +=
@MeantCoder can you please verify if you have /lib/firmware directory in your target? Thanks.

agherzan, 2018-10-03:
@mirzak has the right and detailed answer here. I'll close this issue as this should bring the needed firmware in the image.

closed 2018-10-03 by agherzan