agherzan/meta-raspberrypi #324: omxplayer patch fail on master branch

micha1pu, 2018-10-31:
**Description**

**Steps to reproduce the issue:**
1. bitbake omxplayer
2.
3.

**Describe the results you received:**

omxplayer-git-r4 do_patch: Command Error: 'quilt --quiltrc /home/studio/build/YOCTO_MASTER/poky-thud/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/omxplayer/git-r4/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 0001-Remove-Makefile.include-which-includes-hardcoded.patch
patching file Makefile
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file Makefile
patching file Makefile.ffmpeg
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file Makefile.ffmpeg
The next patch would delete the file Makefile.include,
which does not exist!  Applying it anyway.
patching file Makefile.include
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file Makefile.include
Patch 0001-Remove-Makefile.include-which-includes-hardcoded.patch does not apply (enforce with -f)
ERROR: omxplayer-git-r4 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /home/studio/build/YOCTO_MASTER/poky-thud/build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/omxplayer/git-r4/temp/log.do_patch.17082
ERROR: Task (/home/studio/build/YOCTO_MASTER/poky-thud/meta-raspberrypi/recipes-multimedia/omxplayer/omxplayer_git.bb:do_patch) failed with exit code '1'

**Describe the results you expected:**

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

micha1pu, 2018-10-31:
In latest omxplayer repository file Makefile.include doesn't exist anymore. Does anybody have same problem?

agherzan, 2018-11-01:
This sounds strange. @kraj Could this be introduced by the last update? I haven't tried to reproduce.

kraj, 2018-11-01:
havent seen this here. I will take a look

micha1pu, 2018-11-01:
i just tried to build on clear system with minimal layers and still same problem, on SUMO its working ok.

bitbake omxplayer
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |################################################################################| Time: 0:00:18
Parsing of 1472 .bb files complete (0 cached, 1472 parsed). 2106 targets, 98 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi0-wifi"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:411184bfaa6269bf2926bb2a576c0922958cbbb3"
meta-raspberrypi     = "master:0f14d185d387f59ff9f00365ff314cf51e46db58"
meta-oe              = "master:f1511d254632a34c1deb51f4bf8b8c21e7423f51"

Initialising tasks: 100% |#############################################################################| Time: 0:00:00
Sstate summary: Wanted 425 Found 0 Missed 425 Current 12 (0% match, 2% complete)
NOTE: Executing SetScene Tasks
NOTE: Executing RunQueue Tasks
ERROR: omxplayer-git-r4 do_patch: Command Error: 'quilt --quiltrc /home/studio/build/testing/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/omxplayer/git-r4/recipe-sysroot-native/etc/quiltrc push' exited with 0  Output:
Applying patch 0001-Remove-Makefile.include-which-includes-hardcoded.patch
patching file Makefile
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file Makefile
patching file Makefile.ffmpeg
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file Makefile.ffmpeg
The next patch would delete the file Makefile.include,
which does not exist!  Applying it anyway.
patching file Makefile.include
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- rejects in file Makefile.include
Patch 0001-Remove-Makefile.include-which-includes-hardcoded.patch does not apply (enforce with -f)
ERROR: omxplayer-git-r4 do_patch: Function failed: patch_do_patch
ERROR: Logfile of failure stored in: /home/studio/build/testing/rpi-build/tmp/work/arm1176jzfshf-vfp-poky-linux-gnueabi/omxplayer/git-r4/temp/log.do_patch.12819
ERROR: Task (/home/studio/build/testing/poky/meta-raspberrypi/recipes-multimedia/omxplayer/omxplayer_git.bb:do_patch) failed with exit code '1'
NOTE: Tasks Summary: Attempted 557 tasks of which 546 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /home/studio/build/testing/poky/meta-raspberrypi/recipes-multimedia/omxplayer/omxplayer_git.bb:do_patch
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.

kraj, 2018-11-07:
This should be now fixed. Please reopen if needed.

closed 2018-11-07 by kraj