agherzan/meta-raspberrypi #336: omxplayer build issue

TomMettam, 2018-11-21:
I'm currently unable to build the omxplayer recipe (current master) with yocto.

**Steps to reproduce the issue:**
1. Include omxplayer in IMAGE_INSTALL_append
2. Run build

**Describe the results you received:**

| SubtitleRenderer.cpp:43:3: error: conflicting declaration 'typedef struct EGL_DISPMANX_WINDOW_T EGL_DISPMANX_WINDOW_T'
|  } EGL_DISPMANX_WINDOW_T;
|    ^~~~~~~~~~~~~~~~~~~~~
| In file included from /home/tom/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r4/recipe-sysroot/usr/include/EGL/egl.h:36,
|                  from SubtitleRenderer.h:30,
|                  from SubtitleRenderer.cpp:27:
| /home/tom/yocto/build/tmp/work/cortexa7t2hf-neon-vfpv4-yoe-linux-gnueabi/omxplayer/git-r4/recipe-sysroot/usr/include/EGL/eglplatform.h:118:3: note: previous declaration as 'typedef struct EGL_DISPMANX_WINDOW_T EGL_DISPMANX_WINDOW_T'
|  } EGL_DISPMANX_WINDOW_T;
|    ^~~~~~~~~~~~~~~~~~~~~

**Describe the results you expected:**

omxplayer to successfully compile.

**Additional information you deem important (e.g. issue happens only occasionally):**

This appears to have been introduced with f661406e2df48f1b (0001-Fix-build-with-vc4-driver.patch)

When I revert meta-raspberrypi to 229363bbac the compile succeeds.

kraj, 2018-11-21:
I am not able to reproduce it. but I do have the https://github.com/agherzan/meta-raspberrypi/pull/337
on my tree. Can you try it there. Secondly also post your DISTRO_FEATURES so we can reproduce it.

TomMettam, 2018-11-21:
I shall try this when I get home tonight. My configuration is:

DISTRO_FEATURES_remove_rpi = "x11 wayland"
WPE_PLATFORM = "egl touch"
WPE_BACKEND = "rpi libinput"

with wpe-eglfs-image from meta-wpe

novalisek, 2018-11-21:
Hello,
I am also having same compile problem using those layers:

git clone -b thud git://git.openembedded.org/meta-openembedded
git clone -b thud https://github.com/meta-qt5/meta-qt5
git clone -b thud git://git.yoctoproject.org/meta-security
git clone -b master git://git.yoctoproject.org/meta-raspberrypi
git clone -b sumo git://github.com/jumpnow/meta-rpi

DISTRO_FEATURES = "opengl usbhost ${DISTRO_FEATURES_LIBC}"

bitbake omxplayer

When I manualy delete struct EGL_DISPMANX_WINDOW_T from SubtitleRenderer.cpp, it compiles, but doesn't play anything, when I run omxplayer on RPI3+  

in previous [sumo] branch (of all layers), omxplayer worked fine

I wanted to try new [thud] branch so that I can try if qtmultimedia + gstreamer is working better than in [sumo] (not working IP camera etc ..)

kraj, 2018-11-21:
https://github.com/agherzan/meta-raspberrypi/pull/340 should fix it. Please test and report back

novalisek, 2018-11-21:
I applied #340, then bitbaked omxplayer again and it seems to be working, thanks!

rcla, 2018-11-21:
Is there any possibility to run omxplayer with raspberrypi3-64?

rcla, 2018-11-22:
I have a 64 bit image for rpi3B+

MACHINE ??= "raspberrypi3-64"
DISTRO_FEATURES_append = " opengl alsa x11"
MACHINE_FEATURES_append = " vc4graphics"

bitbake omxplayer

It is failing in the configuration of the ffmpeg package:

ERROR: omxplayer-git-r4 do_compile: oe_runmake failed
ERROR: omxplayer-git-r4 do_compile: Function failed: do_compile (log file is located at /home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/temp/log.do_compile.1279)
ERROR: Logfile of failure stored in: /home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/temp/log.do_compile.1279
Log data follows:
| DEBUG: SITE files ['endian-little', 'bit-64', 'arm-common', 'arm-64', 'common-linux', 'common-glibc', 'aarch64-linux', 'common']
| DEBUG: Executing shell function do_compile
| NOTE: make -j 4 -f Makefile.ffmpeg
| cd ffmpeg; \
| CFLAGS="-D__STDC_CONSTANT_MACROS -D__STDC_LIMIT_MACROS -DTARGET_POSIX -DTARGET_LINUX -fPIC -DPIC -D_REENTRANT -D_HAVE_SBRK -D_LARGEFILE64_SOURCE -DHAVE_CMAKE_CONFIG -DHAVE_VMCS_CONFIG -D_REENTRANT -DUSE_VCHIQ_ARM -DVCHI_BULK_ALIGN=1 -DVCHI_BULK_GRANULARITY=1 -DEGL_SERVER_DISPMANX -D_LARGEFILE_SOURCE -D_LARGEFILE64_SOURCE -D__VIDEOCORE4__ -DGRAPHICS_X_VG=1 -U_FORTIFY_SOURCE -Wall -DHAVE_OMXLIB -DUSE_EXTERNAL_FFMPEG  -DHAVE_LIBAVCODEC_AVCODEC_H -DHAVE_LIBAVUTIL_MEM_H -DHAVE_LIBAVUTIL_AVUTIL_H -DHAVE_LIBAVFORMAT_AVFORMAT_H -DHAVE_LIBAVFILTER_AVFILTER_H -DOMX -DOMX_SKIP64BIT  -D__GBM__                    -isystem/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot/usr/include/interface/vcos/pthreads                    -isystem/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot/usr/include/freetype2                    -isystem/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot/usr/include/interface/vmcs_host/linux                    -isystem/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot/usr/include/dbus-1.0                    -isystem/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot/usr/lib/dbus-1.0/include                   " \
| LDFLAGS="" \
|   ./configure \
| 		--enable-cross-compile \
| 		--ld="aarch64-poky-linux-gcc  --sysroot=/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot" \
| 		--cc="aarch64-poky-linux-gcc  --sysroot=/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot" \
| 		--cxx="aarch64-poky-linux-g++  --sysroot=/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot" \
| 		--extra-cflags="  --sysroot=/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot" \
| 		--extra-ldflags="  --sysroot=/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot" \
| 		--enable-shared \
| 		--disable-static \
| 		--arch=arm \
| 		--cpu=generic \
| 		--target-os=linux \
| 		--disable-hwaccels \
| 		--enable-parsers \
| 		--disable-muxers \
| 		--disable-filters \
| 		--disable-encoders \
| 		--disable-devices \
| 		--disable-programs \
| 		--enable-shared \
| 		--disable-doc \
| 		--disable-postproc \
| 		--enable-gpl \
| 		--enable-version3 \
| 		--enable-protocols \
| 		--disable-libsmbclient \
| 		--enable-libssh \
| 		--enable-nonfree \
| 		--enable-openssl \
| 		--enable-pthreads \
| 		--enable-pic \
| 		--disable-armv5te \
| 		--disable-neon \
| 		--enable-armv6t2 \
| 		--enable-armv6 \
| 		--enable-hardcoded-tables \
| 		--disable-runtime-cpudetect \
| 		--disable-debug \
| 		--disable-crystalhd \
| 		--disable-decoder=h264_vda \
| 		--disable-decoder=h264_crystalhd \
| 		--disable-decoder=h264_vdpau \
| 		--disable-decoder=vc1_crystalhd \
| 		--disable-decoder=wmv3_crystalhd \
| 		--disable-decoder=wmv3_vdpau \
| 		--disable-decoder=mpeg1_vdpau \
| 		--disable-decoder=mpeg2_crystalhd \
| 		--disable-decoder=mpeg4_crystalhd \
| 		--disable-decoder=mpeg4_vdpau \
| 		--disable-decoder=mpeg_vdpau \
[... 164 lines cut ...]
| 		--disable-decoder=v408 \
| 		--disable-decoder=yuv4 \
| 		--disable-decoder=sanm \
| 		--disable-decoder=paf_video \
| 		--disable-decoder=avrn \
| 		--disable-decoder=cpia \
| 		--disable-decoder=bintext \
| 		--disable-decoder=xbin \
| 		--disable-decoder=idf \
| 		--disable-decoder=hevc \
| 		--enable-decoder=opus \
| 		--disable-stripping \
| 		--pkg-config=pkg-config
| aarch64-poky-linux-gcc  --sysroot=/home/ecoi/Downloads/rpi-yocto/rpi64-build/tmp/work/aarch64-poky-linux/omxplayer/git-r4/recipe-sysroot is unable to create an executable file.
| C compiler test failed.
| 
| If you think configure made a mistake, make sure you are using the latest
| version from Git.  If the latest version fails, report the problem to the
| ffmpeg-user@ffmpeg.org mailing list or IRC #ffmpeg on irc.freenode.net.
| Include the log file "ffbuild/config.log" produced by configure as this will help
| solve the problem.
| Makefile.ffmpeg:21: recipe for target 'configure' failed
| make: *** [configure] Error 1
| ERROR: oe_runmake failed
| WARNING: exit code 1 from a shell command.

TomMettam, 2018-11-22:
@rcla Is that relevant to this ticket?

@kraj  - Can confirm, #340 fixes all omxplayer issues for me. Thanks!

rcla, 2018-11-22:
@TomMettam  > @rcla Is that relevant to this ticket?

I think so. Because that happens when I try to compile the omxplayer package:

bitbake omxplayer

TomMettam, 2018-11-22:
@rcla  It's a different issue though, you should post it under a new ticket.

rcla, 2018-11-22:
Ok, then I'll make a new ticket.
Thank you

agherzan, 2018-11-22:
Thanks @TomMettam @kraj . https://github.com/agherzan/meta-raspberrypi/pull/340 Was merged.

closed 2018-11-22 by agherzan