agherzan/meta-raspberrypi #338: No wlan0 while brcmfmac is loaded and brcmfmac43430 installed.

pepijndevos, 2018-11-21:
**Description**

Other issues related to WiFi issues seem to be due to `linux-firmware-brcm43430` not being installed. This is not the case here, as it came included without further action.

The image boots fine, and from dmesg it seems the firmware is uploaded to the wifi chip, yet no `wlan0` interface is present.

**Steps to reproduce the issue:**
1. Execute the steps in the readme
2. set `MACHINE` to `raspberrypi3`
3. Add a new layer with the following image recipe
4. `bitbake` and boot the generated image
5. `ifconfig`: no `wlan0`

```
# Base this image on core-image-base
include recipes-core/images/core-image-base.bb

IMAGE_FEATURES += "ssh-server-dropbear"

IMAGE_INSTALL += "gstreamer networkmanager"
```

I tried adding these lines as mentioned on #31, but without any change

```
DISTRO_FEATURES_append += wifi
IMAGE_INSTALL_append += linux-firmware-brcm43430
```

**Describe the results you received:**

The following is printed in dmesg. Nothing else that seems remotely related to WiFi
 ```
[    2.304635] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    2.311173] brcmfmac: brcmf_fw_map_chip_to_name: using brcm/brcmfmac43430-sdio.bin for chip 0x00a9a6(43430) rev 0x000001
[    2.590453] brcmfmac: brcmf_c_preinit_dcmds: Firmware version = wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[    2.591410] brcmfmac: brcmf_c_preinit_dcmds: CLM version = API: 12.2 Data: 7.11.15 Compiler: 1.24.2 ClmImport: 1.24.1 Creation: 2014-05-26 10:53:55 Inc Data: 9.10.39 Inc Compiler: 1.29.4 Inc ClmImport: 1.36.3 Creation: 2017-10-23 03:47:14 
```

I decided to try a USB WiFi dongle, the one we'll most likely also be using in production (due to external antenna), and that one does not work either.

```
[  192.741298] usb 1-1.2: new high-speed USB device number 6 using dwc_otg
[  192.882649] usb 1-1.2: New USB device found, idVendor=0bda, idProduct=8178
[  192.892649] usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  192.902587] usb 1-1.2: Product: 802.11n WLAN Adapter
[  192.912558] usb 1-1.2: Manufacturer: Realtek
[  192.922400] usb 1-1.2: SerialNumber: 00e04c000001
[  192.933364] rtl8192cu: Chip version 0x11
[  193.013814] rtl8192cu: Board Type 0
[  193.023718] rtl_usb: rx_max_size 15360, rx_urb_num 8, in_ep 1
[  193.033771] rtl8192cu: Loading firmware rtlwifi/rtl8192cufw_TMSC.bin
[  193.044381] ieee80211 phy2: Selected rate control algorithm 'rtl_rc'
[  193.061642] Error: Driver 'rtl8192cu' is already registered, aborting...
[  193.176307] Error: Driver 'rtl8192cu' is already registered, aborting...
```

**Describe the results you expected:**

I expect to see a `wlan0` interface, because the driver seems to load fine and `wpa-supplicant` is also installed.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

```
WARNING: Host distribution "ubuntu-18.04" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |#############################################################################################################################################| Time: 0:00:00
Loaded 3124 entries from dependency cache.
Parsing recipes: 100% |###########################################################################################################################################| Time: 0:00:01
Parsing of 2174 .bb files complete (2173 cached, 1 parsed). 3124 targets, 135 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.40.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi3"
DISTRO               = "poky"
DISTRO_VERSION       = "2.6"
TUNE_FEATURES        = "arm armv7ve vfp thumb neon vfpv4 callconvention-hard cortexa7"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "my-yocto-2.6:84eecb017ef92ef36b4df730908828e54aeff85c"
meta-raspberrypi     = "master:e1ba4484c0135e12fbbd4b7bab00d9a15d5137f2"
meta-oe              
meta-python          
meta-multimedia      
meta-networking      = "mythud:f1511d254632a34c1deb51f4bf8b8c21e7423f51"
meta-rove            = "my-yocto-2.6:84eecb017ef92ef36b4df730908828e54aeff85c"
```

pepijndevos, 2018-11-22:
I feel really stupid now... in my defense, in all the years I've been using normie desktop Linux, I've never had a working interface be *not* up.

So I guess it's not a bug but a feature. Any pointers on how to configure it to come up at boot and connect to WiFi is appreciated. Plan is to use `networkmanager` for automatically reconnecting.

[image]

agherzan, 2018-11-22:
That is fine. If you install and run a network manager (like NetworkManager or conman) the interfaces will be brought up automatically.

closed 2018-11-22 by agherzan

pepijndevos, 2018-11-22:
Thanks a lot. As you can see in my image recipe, I did install NetworkManager, but maybe it's not started by default. I also think that a normal Raspbian light comes with only wpa-supplicant, and that also brings up the interfaces somehow. I'll do some more digging.

agherzan, 2018-11-22:
That is as well true. But you need to start the services as well. I'm not sure how you configured the packages / image.