agherzan/meta-raspberrypi #339: Please make a thud branch

otavio, 2018-11-21:
Hello team,

I think it is a good moment to create the thud branch as people are starting to create projects which rely on this BSP and pinning it to thud makes unsurprising changes more unlikely.

I am not against backporting changes related to firmware rework (#331) and maybe others but using the branch is indeed a good practice.

kraj, 2018-11-21:
I agree however lets wait until we merge the omxplayer fixes.

otavio, 2018-11-21:
It is fine; I just want to don't let it take too long. We can always backport the changes or even keep both branches on sync for some time.

agherzan, 2018-11-22:
I will create it these days. @kraj do you want to take a look at the omxplayer build issue for 64 bit too? i won't be able to look into it soon.

kraj, 2018-11-22:
Omxplayer on 64bit has not been tested here so far I would not hold branching out for that I was more interested to fix it for 32bit with userland graphics since that was a regression