agherzan/meta-raspberrypi #342: gstreamer-1.0 - Bus Error/SIGBUS

novalisek, 2018-11-24:
**Description**

Hello,
I am having problems with gstreamer - when I want to play local video or rtsp stream from IP camera, I always get "Bus error"

May be it has something to do with #319 ?

First I was having this issue with meta-qt5 when using QMediaPlayer, but I found out this seems to be gstreamer error (so I am just reproducing on rpi-basic-image).

I didn't have this problem with [sumo] branch

 

**Steps to reproduce the issue:**

git clone -b thud git://git.yoctoproject.org/poky.git poky-thud
git clone -b thud git://git.openembedded.org/meta-openembedded
git clone -b master git://git.yoctoproject.org/meta-raspberrypi

in local.conf;
IMAGE_INSTALL_append = " connman connman-client gstreamer1.0 gstreamer1.0-plugins-bad gstreamer1.0-plugins-good  gstreamer1.0-plugins-base gstreamer1.0-libav"
MACHINE = "raspberrypi3"

bitbake rpi-basic-image

Then when I start
gst-play-1.0 /opt/video.mp4
or
gst-play-1.0 rtsp://192.168.1.161/0

**Describe the results you received:**

I always get "Bus error"
I am attaching  debug logs (level 4 and 5)
[gst-play-debug4.txt](https://github.com/agherzan/meta-raspberrypi/files/2612503/gst-play-debug4.txt)
[gst-play-debug5.zip](https://github.com/agherzan/meta-raspberrypi/files/2612505/gst-play-debug5.zip)

**Describe the results you expected:**

Video should be played

**Additional information you deem important (e.g. issue happens only occasionally):**
omxplayer is playing video.mp4 without problems, but it has problems with IP camera stream (but I think it is another problem, in omxplayer)

**Additional details (revisions used, host distro, etc.):**

kraj, 2018-11-28:
Please try adding ```gstreamer1.0-omx``` to ```IMAGE_INSTALL_append ``` list.
if that does not work then also add
```
MACHINE_FEATURES_append_rpi = " vc4graphics"
```

and see if that works. 

And make sure you use master or thud branch

kraj, 2018-11-28:
I am using https://github.com/YoeDistro/yoe-distro

bitbake yoe-qt5-wayland-image

and I dont see this issue.

you can also try building core-image-xfce or core-image-x11 if you are using X11 and not wayland

```
root@raspberrypi3:~# gst-play-1.0 https://sample-videos.com/video123/mp4/720/big_buck_bunny_720p_1mb.mp4
[  841.394863] usb 1-1.1.2: reset high-speed USB device number 8 using dwc_otg
Press 'k' to see a list of keyboard shortcuts.
Now playing https://sample-videos.com/video123/mp4/720/big_buck_bunny_720p_1mb.mp4
Prerolling...
WARNING Could not initialise Wayland output
WARNING debug information: ../../../gst-plugins-bad-1.14.4/ext/wayland/gstwaylandsink.c(333): gst_wayland_sink_find_display (): /GstWaylandSin
k:waylandsink0:
Failed to create GstWlDisplay: 'Failed to connect to the wayland display '(default)''
Redistribute latency...
0:00:05.3 / 0:00:05.3
Reached end of play list.
```

novalisek, 2018-11-29:
Hello,
adding gstreamer1.0-omx helped, gst-play is now playing videos (I had to add gpu_mem=128 to config.txt, before that, video was flickering). Ip camera is still not working (you can see attached log  
[gst-play-rtsp.txt](https://github.com/agherzan/meta-raspberrypi/files/2630996/gst-play-rtsp.txt) ), but I think that is for another issue I should put somewhere else.

I also cannot make QMediaPlayer to play the video 

QVideoWidget *video_widget = new QVideoWidget;
QMediaPlayer *player = new QMediaPlayer;
player->setVideoOutput(video_widget);
ui->graphicsView->setViewport(video_widget);
player->setMedia(url_local2);
player->play();

but I will be playing with that ...
If I won't make it work with http://github.com/jumpnow/meta-rpi layer I will try yoe-distro.

Thanks for help.

novalisek, 2018-12-03:
I tried to "bitbake yoe-qt5-eglfs-image" with attached site.conf 
[site.conf.txt](https://github.com/agherzan/meta-raspberrypi/files/2641299/site.conf.txt) - I had to add DISTRO_FEATURES_remove_rpi = "x11 wayland" or it would fail.

When I try to use "omxplayer  big_buck_bunny_720p_1mb.mp4", everything is playing ok, stretched to fullscreen (1920x1080)
When I do 

> root@raspberrypi3:/opt# gst-play-1.0 big_buck_bunny_720p_1mb.mp4
Press 'k' to see a list of keyboard shortcuts.
Now playing /opt/big_buck_bunny_720p_1mb.mp4
Redistribute latency...
0:00:05.3 / 0:00:05.3
Reached end of play list.

the video is flickering and I have to set gpu_mem=128 in config.txt, when I do that, it is playing, but not fullscreen, but that doesn't matter.

When I play another fullHD video (without gpu_mem=128), I get

> gst-play-1.0 video.mp4
Press 'k' to see a list of keyboard shortcuts.
Now playing /opt/video.mp4
Redistribute latency...
eglCreateImageKHR:  failed to create image for buffer 0x2 target 12465 error 0x300c
ERROR Failed to convert video buffer for file:///opt/video.mp4
ERROR debug information: ../../../gst-plugins-base-1.14.4/ext/gl/gstglcolorconvertelement.c(218): gst_gl_color_convert_element_prepare_output_buffer (): /GstPlayBin:playbin/GstPlaySink:playsink/GstBin:vbin/GstGLImageSinkBin:glimagesinkbin0/GstGLColorConvertElement:glcolorconvertelement0
Reached end of play list.

when I add gpu_mem=128, it is playing, but flickering 
I have to use  gpu_mem_256=192 in config.txt (and it is still loosing some frames I think)
omxplayer is playing this video fine without any changes in config.txt 

So I think there are still some problems in gstreamer, but that may be for another bug report somewhere else (don't know where now)

I think this issue can be closed (there is no bus error anymore)

kraj, 2018-12-03:
if you have 1GB RAM then you can try allocating say 392 to gpu_mem and see if that helps.

novalisek, 2018-12-03:
yes, I figured it out, but I was just curious why there are no changes needed with omxplayer ... 
I will try if qMediaplayer works after I bitbake SDK, but first I will have to resize my virtual disk :) because 80GB is not enough for "bitbake yoe-qt5-eglfs-image" + "bitbake meta-toolchain-qt5"   (45GB was enough for  qt5-basic-image + meta-toolchain-qt5 from  https://github.com/jumpnow/meta-rpi, it is strange that it is so big difference )

kraj, 2018-12-03:
Well ```yoe-qt5-eglfs-image``` contains all QT5 examples since its a reference image. you can remove them if you want. Just set ```QT5APPS_forcevariable = ""``` in ```local.conf```

novalisek, 2018-12-03:
will try, thank you. Is there any way to delete/cleanup space occupied by those examples? Or only to find out their names somehow and bitbake -c clean packagenames?

kraj, 2018-12-03:
Just delete tmp/ dir and rebuild.

novalisek, 2018-12-03:
It will take ages to rebuild all again. I was thinking about something faster. But I can turn off the radiator valves for a day :)

kraj, 2018-12-03:
it wont. if you have built it once. It will reuse shared state and rebuilt only the things that are not built in previous builds.

novalisek, 2018-12-03:
ok, I will try that, still learning :+1:

closed 2018-12-03 by novalisek