agherzan/meta-raspberrypi #343: How to add gstreamer1.0-plugins-ugly to meta-raspberrypi

madhu2051, 2018-11-27:
I want to play the mp3 files from Raspberry PI. While playing an mp3 song, I found that there is no mp3 plugin on my raspberry system.

How can I add the gstreamer-plugin-ugly to meta-raspberrypi recipe?

Regards,
-Madhu.

kraj, 2018-11-28:
In local.conf add

```
IMAGE_INSTALL_append = " gstreamer1.0-plugins-good gstreamer1.0-plugins-base gstreamer1.0-plugins-ugly"
LICENSE_FLAGS_WHITELIST_append = " commercial_gstreamer1.0-plugins-ugly "

```

Then bitbake your image. Flash it and then play mp3

```
$ gst-play-1.0 foo.mp3
```

madhu2051, 2018-11-28:
Hi Kraj,
           After adding the above lines in build/conf/local.conf, I am getting the following errors while compilation.

/poky/meta/recipes-multimedia/gstreamer/gstreamer1.0-plugins-ugly_1.12.2.bb DEPENDS on or otherwise requires it)
mpg123 was skipped: because it has a restricted license not whitelisted in LICENSE_FLAGS_WHITELIST
NOTE: Runtime target 'gstreamer1.0-plugins-ugly' is unbuildable, removing...
Missing or unbuildable dependency chain was: ['gstreamer1.0-plugins-ugly', 'mpg123']
ERROR: Required build target 'agl-demo-platform' has no buildable providers.

kraj, 2018-11-28:
also add 
```
LICENSE_FLAGS_WHITELIST_append = " commercial_mpg123 "
```

madhu2051, 2018-12-03:
Thanks Kraj, Issue resolved.

closed 2018-12-03 by madhu2051