agherzan/meta-raspberrypi #346: I2C module not loaded after enable it

elct9620, 2018-12-01:
**Description**

The i2c is enabled by `rpi-conifg` but it didn't show inside `/dev/i2c-*`, after add `/etc/modules-load.d/i2c-dev.conf` with content `i2c-dev` and it is shown.

**Steps to reproduce the issue:**
1. add `ENABLE_I2C="1"` in `conf/local.conf`
2. build rpi3 image
3. boot rpi3 use this generated image

**Describe the results you received:**

The `lsmod` can found `i2c_bcml2835` but `i2c_dev` is not showing, and `/dev/i2c-*` is not exists.

**Describe the results you expected:**

The config may not necessary by default, but it seems to be an option to auto add `/etc/modules-load.d/i2c-dev.conf` to enable `i2c_dev` feature.

**Additional information you deem important (e.g. issue happens only occasionally):**

**Additional details (revisions used, host distro, etc.):**

agherzan, 2018-12-01:
The config that you mention enables i2c correctly. It doesn't mean it exposes the userspace adapter. Generally, the i2c devices are controlled via kernel modules functionality being enabled with ENABLE_I2C.

elct9620, 2018-12-01:
So the `ENABLE_I2C` only enable kernel modules, if I need `/dev/i2c-*` to the controller it, I have to add some config into `/etc/modules-load.d/i2c-dev.conf` to expose it. And it should not as a default behavior.

I am not sure which is the normal way to use it. So this makes me confused about this config. I use ruby direct to interactive with the i2c interface, so I think the `/dev/i2c-*` will exists by default.

asmaaswt, 2018-12-01:
now it's working for host fine

agherzan, 2018-12-04:
@asmaaswt I don't understand your comment. Could you develop?
@elct9620 That is the intended behaviour. The module load nor i2c dev functionality is RaspebrryPi specific and depends very much on your setup. If you want to use i2c userspace adapter you will need to load the i2c dev module. The BSP provides the functionality underneath.

elct9620, 2018-12-04:
@agherzan Thanks. This is the answer I want to check for it.

closed 2018-12-04 by elct9620

agherzan, 2018-12-04:
Thanks, @elct9620 .