agherzan/meta-raspberrypi #349: valgrind

TomMettam, 2018-12-03:
Valgrind from oe-core does not work correctly on Raspberry Pi 3 (armv7)

Could this be fixed by this overlay?

1.  Add valgrind to your recipe
2.  Build & Deploy
3.  Run Valgrind against a simple app
4. Valgrind's output does not include any line information

Simple test program:

`#include <stdlib.h>

int main(int argc, char **argv)
{
        void *foo=malloc(10);
        return 0;
}`

Output with yocto's valgrind:

`root@raspberrypi3:~/leak# gcc -ggdb leak.c
root@raspberrypi3:~/leak# valgrind --leak-check=full ./a.out
==20186== Memcheck, a memory error detector
==20186== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==20186== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==20186== Command: ./a.out
==20186==
==20186==
==20186== HEAP SUMMARY:
==20186==     in use at exit: 10 bytes in 1 blocks
==20186==   total heap usage: 1 allocs, 0 frees, 10 bytes allocated
==20186==
==20186== 10 bytes in 1 blocks are definitely lost in loss record 1 of 1
==20186==    at 0x483E970: malloc (in /usr/lib/valgrind/vgpreload_memcheck-arm-linux.so)
==20186==
==20186== LEAK SUMMARY:
==20186==    definitely lost: 10 bytes in 1 blocks
==20186==    indirectly lost: 0 bytes in 0 blocks
==20186==      possibly lost: 0 bytes in 0 blocks
==20186==    still reachable: 0 bytes in 0 blocks
==20186==         suppressed: 0 bytes in 0 blocks
==20186==
==20186== For counts of detected and suppressed errors, rerun with: -v
==20186== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)`

After downloading and compiling the same version manually on the target:

`root@raspberrypi3:~/leak# valgrind --leak-check=full ./a.out
==790== Memcheck, a memory error detector
==790== Copyright (C) 2002-2017, and GNU GPL'd, by Julian Seward et al.
==790== Using Valgrind-3.14.0 and LibVEX; rerun with -h for copyright info
==790== Command: ./a.out
==790==
==790==
==790== HEAP SUMMARY:
==790==     in use at exit: 10 bytes in 1 blocks
==790==   total heap usage: 1 allocs, 0 frees, 10 bytes allocated
==790==
==790== 10 bytes in 1 blocks are definitely lost in loss record 1 of 1
==790==    at 0x483E604: malloc (vg_replace_malloc.c:299)
==790==    by 0x108527: main (leak.c:5)
==790==
==790== LEAK SUMMARY:
==790==    definitely lost: 10 bytes in 1 blocks
==790==    indirectly lost: 0 bytes in 0 blocks
==790==      possibly lost: 0 bytes in 0 blocks
==790==    still reachable: 0 bytes in 0 blocks
==790==         suppressed: 0 bytes in 0 blocks
==790==
==790== For counts of detected and suppressed errors, rerun with: -v
==790== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
`

Notice that the line refernece to leak.,c is now available

kraj, 2018-12-04:
This is happening because ```/usr/lib/valgrind/vgpreload_memcheck-arm-linux.so``` is stripped of its symbols, in order for us to get useful backtraces its must that this shared object from valgrind is not stripped. So how do we do this 

1. install ```valgrind-dbg``` into image. if you have feeds then simply doing ```opkg install valgrind-dbg``` , this might be big but I guess on rpi you should be ok, just use a bigger SD card.

2. Do not strip valgrind, we can do this by adding ```INHIBIT_PACKAGE_STRIP = “1” ``` in valgrind recipe

regardless, I think we can improve the situation by letting valgrind recipe to not strip this particular shared object since its special. Then we dont need to include full dbg package in image. This however is OE-Core problem, feel free to open a ticket in yocto project bugzilla. I might get to it eventually
unless someone picks it up and fixes before that

kraj, 2018-12-15:
Can we close this now ? the patch to fix this is already in master in core

http://git.openembedded.org/openembedded-core/commit/?id=23da8f50b1e0a74777035c9f7b65b81456908f9f

closed 2018-12-15 by kraj