agherzan/meta-raspberrypi #355: Gstreame-plugings-(bad|good) fail to install after updated

pepijndevos, 2018-12-12:
**Description**

I have a recipe that RDEPENDS on `gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-bad gstreamer1.0-omx`, but when I updated to the latest master in the hope to fix a [firmware issue with omx](https://github.com/raspberrypi/firmware/issues/1051), half of the plugins stopped being included. The base and omx plugins are still there, but all the good/bad ones are missing.

**Steps to reproduce the issue:**
Build an image that should include `gstreamer1.0-plugins-good` using the latest master.
Look in `/usr/lib/gstreamer1.0` in the sysroot or final image to check it was actually included.

**Describe the results you received:**

The recipe builds fine, and the files are there inside its image folder. It also generates the RPM just fine. The recipe is also on the manifest of the final image. Except the shared libraries aren't there.

**Describe the results you expected:**

There should be a whole bunch of shared libraries in `/usr/lib/gstreamer1.0`, but after updating `meta-raspberrypi` to the latest master, they are no longer there.

**Additional information you deem important (e.g. issue happens only occasionally):**

I've tried updating poky and openembed, with no change. I've run a git bisect, and the offending commit is a6f96854b61fa8e512c30d9ca814087e0d7822f8, which I don't really understand why.

**Additional details (revisions used, host distro, etc.):**

I was on e1ba448, and updating to c8a05f2 broke it. I was on poky tag yocto-2.6, but I'm now on master without any change. My host is Ubuntu 18.04.

p.s. not sure if the firmware was recently updated, but that would also be needed to fix the linked OMX issue.

agherzan, 2018-12-12:
@kraj will probably know what happened here.

kraj, 2019-02-28:
@pepijndevos this notification missed me. Are you using 32bit+userland-graphics or is it mesa/vc4graphics stack?

kraj, 2019-02-28:
I see this when above packages are added to `IMAGE_INSTALL`

```
root@raspberrypi3:/usr/lib/gstreamer-1.0# ls
libgstaccurip.so                libgstaudiotestsrc.so           libgstdtmf.so                   libgstgoom.so                   libgstmms.so                    libgstplayback.so               libgstsmpte.so                  libgstvideofilter.so
libgstadder.so                  libgstaudiovisualizers.so       libgstdvb.so                    libgstgoom2k1.so                libgstmpegpsdemux.so            libgstpng.so                    libgstsndfile.so                libgstvideofiltersbad.so
libgstadpcmdec.so               libgstauparse.so                libgstdvbsuboverlay.so          libgsthls.so                    libgstmpegpsmux.so              libgstpnm.so                    libgstsoup.so                   libgstvideoframe_audiolevel.so
libgstadpcmenc.so               libgstautoconvert.so            libgstdvdspu.so                 libgsticydemux.so               libgstmpegtsdemux.so            libgstproxy.so                  libgstspectrum.so               libgstvideomixer.so
libgstaiff.so                   libgstautodetect.so             libgsteffectv.so                libgstid3demux.so               libgstmpegtsmux.so              libgstpulseaudio.so             libgstspeed.so                  libgstvideoparsersbad.so
libgstalaw.so                   libgstavi.so                    libgstencoding.so               libgstid3tag.so                 libgstmpg123.so                 libgstrawparse.so               libgstspeex.so                  libgstvideorate.so
libgstalpha.so                  libgstbayer.so                  libgstequalizer.so              libgstimagefreeze.so            libgstmulaw.so                  libgstremovesilence.so          libgststereo.so                 libgstvideoscale.so
libgstalphacolor.so             libgstbluez.so                  libgstfaad.so                   libgstinter.so                  libgstmultifile.so              libgstreplaygain.so             libgstsubenc.so                 libgstvideosignal.so
libgstalsa.so                   libgstbz2.so                    libgstfaceoverlay.so            libgstinterlace.so              libgstmultipart.so              libgstrfbsrc.so                 libgstsubparse.so               libgstvideotestsrc.so
libgstapetag.so                 libgstcairo.so                  libgstfbdevsink.so              libgstinterleave.so             libgstmxf.so                    libgstrsvg.so                   libgsttaglib.so                 libgstvmnc.so
libgstapp.so                    libgstcamerabin.so              libgstfestival.so               libgstipcpipeline.so            libgstnavigationtest.so         libgstrtp.so                    libgsttcp.so                    libgstvolume.so
libgstasfmux.so                 libgstcoloreffects.so           libgstfieldanalysis.so          libgstisomp4.so                 libgstnetsim.so                 libgstrtpmanager.so             libgsttheora.so                 libgstvorbis.so
libgstaudiobuffersplit.so       libgstcompositor.so             libgstflac.so                   libgstivfparse.so               libgstogg.so                    libgstrtponvif.so               libgsttimecode.so               libgstwavenc.so
libgstaudioconvert.so           libgstcoreelements.so           libgstflv.so                    libgstivtc.so                   libgstomx.so                    libgstrtsp.so                   libgstttmlsubs.so               libgstwavparse.so
libgstaudiofx.so                libgstcurl.so                   libgstflxdec.so                 libgstjp2kdecimator.so          libgstopengl.so                 libgstsbc.so                    libgsttypefindfunctions.so      libgstwebp.so
libgstaudiofxbad.so             libgstcutter.so                 libgstfreeverb.so               libgstjpeg.so                   libgstopenglmixers.so           libgstsdpelem.so                libgstudp.so                    libgsty4mdec.so
libgstaudiolatency.so           libgstdashdemux.so              libgstfrei0r.so                 libgstjpegformat.so             libgstopus.so                   libgstsegmentclip.so            libgstuvch264.so                libgsty4menc.so
libgstaudiomixer.so             libgstdebug.so                  libgstgaudieffects.so           libgstlame.so                   libgstopusparse.so              libgstshapewipe.so              libgstvcdsrc.so                 libgstyadif.so
libgstaudiomixmatrix.so         libgstdebugutilsbad.so          libgstgdkpixbuf.so              libgstlegacyrawparse.so         libgstossaudio.so               libgstshm.so                    libgstvideo4linux2.so
libgstaudioparsers.so           libgstdecklink.so               libgstgdp.so                    libgstlevel.so                  libgstpango.so                  libgstsiren.so                  libgstvideobox.so
libgstaudiorate.so              libgstdeinterlace.so            libgstgeometrictransform.so     libgstmatroska.so               libgstpbtypes.so                libgstsmooth.so                 libgstvideoconvert.so
libgstaudioresample.so          libgstdtls.so                   libgstgio.so                    libgstmidi.so                   libgstpcapparse.so              libgstsmoothstreaming.so        libgstvideocrop.so
```

pepijndevos, 2019-02-28:
I'm no longer at the job where I ran into this problem so I cannot check, sorry.

agherzan, 2019-02-28:
I'll close it for now in that case. If anyone faces this again, feel free to reopen.

closed 2019-02-28 by agherzan