agherzan/meta-raspberrypi #359: U-boot not working on Rapsberry Pi compute module 3 lite on sumo

erikhh, 2018-12-14:
**Description**
My build worked fine on a regular raspberry pi 3. But it produces a rainbow screen when trying to boot it on a Compute Module 3 lite.

**Steps to reproduce the issue:**
The contents of my `local.conf`:

```
# Yocto
DISTRO ?= "poky"
PACKAGE_CLASSES ?= "package_deb"
EXTRA_IMAGE_FEATURES ?= "debug-tweaks"
USER_CLASSES ?= "buildstats image-mklibs image-prelink"
PATCHRESOLVE = "noop"

BB_DISKMON_DIRS ??= "\
    STOPTASKS,${TMPDIR},1G,100K \
    STOPTASKS,${DL_DIR},1G,100K \
    STOPTASKS,${SSTATE_DIR},1G,100K \
    STOPTASKS,/tmp,100M,100K \
    ABORT,${TMPDIR},100M,1K \
    ABORT,${DL_DIR},100M,1K \
    ABORT,${SSTATE_DIR},100M,1K \
    ABORT,/tmp,10M,1K"

CONF_VERSION = "1"

# Mender
INHERIT += "mender-full"

MACHINE = "raspberrypi-cm3"

RPI_USE_U_BOOT = "1"
ENABLE_UART = "1"

MENDER_PARTITION_ALIGNMENT = "4194304"
MENDER_BOOT_PART_SIZE_MB = "40"
MENDER_STORAGE_TOTAL_SIZE_MB = "6144"

IMAGE_INSTALL_append = " kernel-image kernel-devicetree"
IMAGE_FSTYPES_remove += " rpi-sdimg"

MENDER_DEMO_HOST_IP_ADDRESS = "192.168.2.1"

DISTRO_FEATURES_append = " systemd"
VIRTUAL-RUNTIME_init_manager = "systemd"
DISTRO_FEATURES_BACKFILL_CONSIDERED = "sysvinit"
VIRTUAL-RUNTIME_initscripts = ""

ARTIFACTIMG_FSTYPE = "ext4"
SDIMG_ROOTFS_TYPE = "ext4"
```

**Describe the results you received:**

Rainbow screen when booting the image on a Compute Module 3 lite.

**Describe the results you expected:**

A image that boots on my compute module 3 lite board.

**Additional information you deem important (e.g. issue happens only occasionally):**

I managed to get it working by adding this to my `local.conf`:
```
RPI_EXTRA_CONFIG = '\nkernel=kernel7.img\n'
```
My guess is that the Compute Module 3 Lite does not look for kernel7.img by default like the regular Raspberry Pi 3 does.

Since this is a solid solution, maybe the fix is simply documentation, I don't know.

**Additional details (revisions used, host distro, etc.):**

Everything, Mender, Yocto, etc is on sumo.

agherzan, 2018-12-14:
Hi. Can you try this on thud/master?

erikhh, 2018-12-14:
On thud I can't reproduce this behaviour. So I'll close this issue.

closed 2018-12-14 by erikhh