agherzan/meta-raspberrypi #364: Xfce desktop,  thud or master branch

rcla, 2018-12-18:
I have a raspberry pi 3B+ and I wish to have an image with the xfce desktop. (meta-xfce)

It works with the sumo branch, both in 32 and 64 bits. 

But now that I update the thud or master branch, it compiles ok but it does not work well:
When executing the xfce desktop, the panels do not display the edges of the forms and buttons.
It takes a long time to visualize the background. Does not let in the Xfce-terminal.
When you try to log out, it does not respond... that is, many things work erratically.

To produce this issue, just try to create **core-image-minimal-xfce** in the thud or master branch.

I have performed the following tests, with the same erratic performance:

- A 64-bit image (raspberrypi3-64) and meta-xfce with poky-yocto using all the thud branch.
- A 64-bit image (raspberrypi3-64) and meta-xfce with poky-yocto using all the master branch.
- A 64-bit image (raspberrypi3-64) and meta-xfce with yoe-distro using all the master branch.
- A 64-bit image (raspberrypi3-64) and meta-xfce with poky-yocto, meta-raspi-light using all the master branch.
- Currently I tested a 32-bit image (raspberrypi3) and meta-xfce with poky-yocto using all the master branch.

The xfce desktop, only works well if I use the sumo branch in all.

agherzan, 2019-03-13:
It would be worth giving a try to the current HEAD of thud.

schnitzeltony, 2019-03-13:
On Tue, Dec 18, 2018 at 5:07 PM rcla <notifications@github.com> wrote:
> I have a raspberry pi 3B+ and I wish to have an image with the xfce desktop. (meta-xfce)
>
> It works with the sumo branch, both in 32 and 64 bits.
>
> But now that I update the thud or master branch, it [...]
Am not using core-image-minimal-xfce so I don't know what might miss
there but I have XFCE on rasp/imx6 (thud / 32 bits) running perfectly
fine.

* For the formatting panels/buttons issues: I think it is caused by
themes: Huge parts of XFCE moved to GTK3 but XFCE GTK3 themes are in
terrible condition (think XFCE stopped taking care due to GNOME
changing theme format frequently) . So I suggest you add
gnome-theme-adwaita(-dark) and select it in settings/appearence.
* Last time I looked into, I saw that 64Bit has issues. Don't remember
exactly but I considered them caused by X-server and since I do not
use 64Bit myself I did not look into further.
* For master images I strongly recommend to enable 'polkit' in distro-features.

Hope that helps a bit

Andreas

rcla, 2019-03-20:
I have re-created core-image-minimal-xfce with the current thud, but the same issues continue.

schnitzeltony, 2019-04-02:
Seems that this is not an meta-raspberrypi issue (not even an XFCE issue). I changed the audience and added some findings [1]. 
Close this one?

[1] http://lists.openembedded.org/pipermail/openembedded-devel/2019-April/199010.html

agherzan, 2019-04-16:
I agree.

closed 2019-04-16 by agherzan