agherzan/meta-raspberrypi #372: No screen found.

muddessir, 2019-01-21:
RPI: 3 Model B+
Yocto: 2.5.1 sumo 
Machine: raspberrypi3 (_tried also raspberrypi3-64_)
Distro: poky
Local.conf: https://pastebin.com/edit/30rVApxW
Xorg.0.log: https://pastebin.com/raw/4cYZFLns

When I build yocto default with meta-raspberry, after burn image: the device won't start with X11. It says: **(EE) No screen found**.
I read this page (https://github.com/agherzan/meta-raspberrypi/issues/228) Also tried to build both raspberrypi3 and raspberrypi3-64 the result was same.

So May you advice me anything to understand what is the problem?
- Should I install additional Graphic Driver?
- Or Should I write some settings?

agherzan, 2019-01-21:
Hi. Can you give thud a try?

agherzan, 2019-01-22:
Hi @muddessir, as you could see in the other thread, this is known to happen on older branches and because they are not actively maintained we haven't invested time to figure out what are the changes in the thud branch that makes this work. That being said we have two options:
1. you either work on an actively maintained branch (thud)
2. you try to investigate what we miss - I suspect some patches on userland

muddessir, 2019-01-25:
Ok.
I will try to upgrade my system to thud. Afterwards I will try.

Thank you.

agherzan, 2019-03-13:
@muddessir Any updates on this?

muddessir, 2019-03-16:
> @muddessir Any updates on this?

it seems ok with thud.

closed 2019-03-16 by muddessir

kaiwan, 2023-03-12:
Hi,
Am facing the same issue, with a recent ver of dunfell... 
My env:

```
DISTRO:poky
DISTRO_NAME:Poky (Yocto Project Reference Distro)
DISTROOVERRIDES:poky
DISTRO_FEATURES:acl alsa argp bluetooth ext2 ipv4 ipv6 largefile pcmcia usbgadget usbhost wifi xattr nfs zeroconf pci 3g nfc x11 vfat largefile opengl ptest multiarch   x11 fbdev directfb opengl pulseaudio sysvinit gobject-introspection-data ldconfig
DISTRO_VERSION = "3.1.23"
DISTRO_CODENAME = "dunfell"
INIT manager:sysvinit
BB_VERSION           = "1.46.0"
TARGET_SYS           = "aarch64-poky-linux"
MACHINE              = "raspberrypi4-64"

```
This is what I see when I run `startx` on the (ssh) console:
```root@raspberrypi4-64:~# startx
xauth:  file /home/root/.serverauth.685 does not exist

X.Org X Server 1.20.14
X Protocol Version 11, Revision 0
Build Operating System: Linux 
Current Operating System: Linux raspberrypi4-64 5.4.72-v8 #1 SMP PREEMPT Mon Oct 19 11:12:20 UTC 2020 aarch64
Kernel command line: coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1  smsc95xx.macaddr=E4:5F:01:49:32:B5 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait    fbcon=map:10 fbcon=font:VGA8x8
Build Date: 15 December 2021  07:01:48PM
 
Current version of pixman: 0.38.4
	Before reporting problems, check http://wiki.x.org
	to make sure that you have the latest version.
Markers: (--) probed, (**) from config file, (==) default setting,
	(++) from command line, (!!) notice, (II) informational,
	(WW) warning, (EE) error, (NI) not implemented, (??) unknown.
(==) Log file: "/var/log/Xorg.0.log", Time: Fri Mar  9 18:41:42 2018
(==) Using config file: "/etc/X11/xorg.conf"
(==) Using config directory: "/etc/X11/xorg.conf.d"
(==) Using system config directory "/usr/share/X11/xorg.conf.d"
(EE) 
Fatal server error:
(EE) no screens found(EE) 
(EE) 
Please consult the The X.Org Foundation support 
	 at http://wiki.x.org
 for help. 
(EE) Please also check the log file at "/var/log/Xorg.0.log" for additional information.
(EE) 
(EE) Server terminated with error (1). Closing log file.
```

I have a Raspberry Pi 4 with a generic 3.5" LCD TFT display panel attached...
I've specified the foll in my local.conf:
``` 
MACHINE_FEATURES_append = " pitft pitft35r vc4graphics"
DISTRO_FEATURES_append = " x11 fbdev directfb opengl"
DISTRO_FEATURES_remove = " wayland vulkan"
```
Is this appropriate for the display? Am I missing something (or am I specifying too many?); Do we need fbdev, directfb?
Or is just 'x11' sufficient? This is something that's quite unclear...

Any help's appreciated! Thanks in advance,

EasyIP2023, 2023-03-12:
@mehmetserter Is the corresponding DRM driver for Ilitek driver chip based displays loaded as module or built into kernel. That's where I'd start. ILI9488 driver chip DRM driver is purposed here https://lore.kernel.org/all/cover.1592055494.git.kamlesh.gurudasani@gmail.com/. This may be of use to you.

I believe how stock Xorg config works is it utilizes the character device node associated with the VC4 DRM driver when vc4graphics is enabled first, then tries other options such as old school fbdev. Make sure to include the corresponding Xorg driver to get it to work properly https://gitlab.freedesktop.org/xorg/driver.

All known kernel ilitek DRM drivers
https://elixir.bootlin.com/linux/v6.2.5/source/drivers/gpu/drm/tiny

All I got for you

EasyIP2023, 2023-03-12:
Also, @kaiwan Instead of `startx` try.

```
/usr/bin/Xorg :0 -s 0 -dpms -noreset -nolisten tcp -nocursor -background none
```

kaiwan, 2023-03-12:
@EasyIP2023 , thanks... I did make some progress, your inputs help, but it isn't working, yet...

From https://github.com/goodtft/LCD-show , I installed stuff as shown via the LCD35-show script.
This helps bringup the graphics display but in console mode only... (fbcon i guess):
```
root@raspberrypi4-64:~# lsmod|grep fb     
fb_ili9486             16384  1
fbtft                  40960  1 fb_ili9486
```
 Xorg doesn't run successfully, same error
```
(EE) 
Fatal server error:
(EE) no screens found(EE) 
(EE) 
```
...Even with the variation you suggested...
How come? Anything you'll can think of?

TIA!

EasyIP2023, 2023-03-13:
@kaiwan

There should be three more things to change/add

* Add appropriate device tree overlay to `/boot/overlays`. More that likely found in kernel build output `arch/arm/boot/dts`
* Update `config.txt` to include `dtparam=spi=on` and `dtoverlay=<device tree overlay name>`.
* Update Xorg config so that is uses character device associated with fb_ili9486 driver

I'm assuming the chosen communication protocol to interact with display is SPI. Check this link out if you haven't already for more info

https://www.willprice.dev/2017/09/16/adventures-with-tft-screens-for-raspberry-pi.html

kaiwan, 2023-03-13:
Thanks! Busy now, will check this out...