agherzan/meta-raspberrypi #375: raspberrypi0w - no wifi interface on thud branch

nandra, 2019-01-24:
Build image for raspberrypi0w  from thud branch. After startup check ifconfig -a and no wlan0 interface was present. Anyway FW files are present:
```
ls /lib/firmware/brcm/
BCM43430A1.hcd          brcmfmac43430-sdio.bin  brcmfmac43430-sdio.txt
```
**Steps to reproduce the issue:**
1. build image from thud branch for rpi0w

**Describe the results you received:**
No wlan0 interface available

**Describe the results you expected:**
wlan0 interface should be working

agherzan, 2019-01-25:
Hi @nandra . Can you attach the `dmesg` log?

nandra, 2019-01-25:
Sure: https://pastebin.com/AG7YVdVD

+ I also enabled u-boot for rpi (needed for rauc update)

Looks like wifi driver is not probed at all.

nandra, 2019-01-30:
@agherzan any ideas? I know for sure that on rocko ifi was working. I can bisect maybe but it will take some time ;)

agherzan, 2019-01-30:
@nandra That is strange indeed. I'd have to take a look (but that will happen after I get rid of this annoying cold).

kraj, 2019-01-30:
@nandra what does `ifconfig -a` says

nandra, 2019-01-30:
@kraj shows only lo interface nothing more (lsmod also doesn't report any wifi drivers available while bluetooth driver is loaded).

kraj, 2019-01-30:
that means its not getting right firmware in image, can you try to manually copy the /lib/firmware directory from  a working SD card to thud and see if that helps.

agherzan, 2019-01-30:
@kraj But that should have been shown in dmesg, isn't it?

kraj, 2019-01-30:
@agherzan good point, (with another sip of morning tea) I looked at dmesg again and I don't see it loading `brcmfmac` module which means the wifi chip will go undetected, can we find out why this module is not being loaded ?

nandra, 2019-01-30:
Asked for help also here as it looks like kernel issue: https://github.com/raspberrypi/linux/issues/2842

nandra, 2019-01-31:
Looks like wifi chip became dead somehow (pls see info here:) https://github.com/raspberrypi/linux/issues/2842

Basically tried raspbian and interface is also not preset.

closed 2019-01-31 by nandra

agherzan, 2019-01-31:
@nandra can you retry the version you know for sure it was working? We found something similar with a testing device lately.

nandra, 2019-01-31:
@agherzan ok building image now and will do the shot

nandra, 2019-01-31:
@agherzan I flashed image for whcih I have confirmed wifi works on other rpi0w but not on mine. So I think my HW is really dead. Can you try raspbian if wifi will work there? Thanks.

agherzan, 2019-02-01:
@nandra I actually did and it is reproducible but it seems like a different issue (maybe?): https://github.com/RPi-Distro/repo/issues/122

nandra, 2019-02-01:
@agherzan  Hmm in my log there are no such issues. So also in raspbian image you don't have wifi interface? I had yesterday confirmation that image build on rocko wifi works perfectly fine. Weird :)

agherzan, 2019-02-01:
That is correct, It fails as well in raspbian for this OTG + Wifi dongle setup. But only when rebooting for a couple of times.

nandra, 2019-02-01:
So for first few time everything is fine but then later after reboot make troubles. Maybe ask on linux-raspberrypi github

agherzan, 2019-02-01:
That's the plan. For now, I'll let the raspbian guys know.

nandra, 2019-02-04:
Looks like also raspbian have an issue: https://github.com/raspberrypi/linux/issues/2846

nandra, 2019-02-06:
Also pls note: https://github.com/raspberrypi/linux/issues/2846#issuecomment-460177049